don/cordova-plugin-ble-central #28: rfDuino example not working with BLE Shield

johncban, 2015-01-01:
Good Day,

I use the rfduino example for a BLE shield test: http://wiki.aprbrother.com/wiki/BLEShield 
I change the service UUID, receiveCharacteristic and sendCharacteristic of the said example but after I uploaded or tested to my Note 5 it re forward me to the first screen after tapping the Disconnect/Connect Button. 

Code Details:

var rfduino = {
    serviceUUID: "0000FFF0-0000-1000-8000-00805F9B34FB", 
    receiveCharacteristic: "0000FFF1-0000-1000-8000-00805F9B34FB",
    sendCharacteristic: "0000FFF2-0000-1000-8000-00805F9B34FB"
};

function parseAdvertisingData(bytes) {
    var length, type, data, i = 0, advertisementData = {};

    while (length !== 0) {
    
       length = bytes[i] & 0xFFF1;         
        i++;
        
        type = bytes[i] & 0xFFF2;
        i++;    
        
        data = bytes.slice(i, i + length - 2); // length includes type and length
        i += length - 2;  // move to end of data
        i++;

        advertisementData[type] = data;
    }
    return advertisementData;
}

// RFduino advertises the sketch its running in the Manufacturer field 0xFF
// RFduino provides a UART-like service so all sketchs look the same
var getRFduinoService = function(scanRecord) {
    var ad = parseAdvertisingData(scanRecord),
        mfgData = ad[0xFFF],
        service;
    
    if (mfgData) {
      // ignore 1st 2 bytes of mfg data
      service = bytesToString(mfgData.slice(2));
      return service;      
    } else {
      return "";
    }
};

Is there any other way to skip the parseAdvertisingData function or send any serial command to Arduino without the parseAdvertisingData? 

Thanks in advance and respectfully,
Jc

don, 2015-01-02:
RFduino includes extra advertising info in the Manufacturer Data, which won't apply to your hardware. The [Adafruit](https://github.com/don/cordova-plugin-ble-central/tree/master/examples/bluefruitle) or [RedBearLab](https://github.com/don/cordova-plugin-ble-central/tree/master/examples/redbearlab) examples are probably a better place to start.

johncban, 2015-01-02:
Thanks for the reply Mr. Don, I actually recently purchased this: http://store.redbearlab.com/products/ble-nano

I would like to upgrade to BLE 4 from Blutooth sparkfun for a personal project of mine and I would like to integrate it to the heart rate sensor connected to my arduino. I use your bluetooth plugin a lot for testing and project deployment as I choose cordova/phonegap because of mobile OS flexibility. 

Mr. Don, thank you very much for your help and have a great new year.

Respectfully,
Jc

closed 2015-01-05 by don