don/cordova-plugin-ble-central #34: PGB shows no plugin

JohnHigbie, 2015-01-27:
Hi Don, Added the gap:plugin line to config.sys (and using the chat example for testing), but PGB shows no plugins. Is there something else I need to do here?
Thanks, John

don, 2015-01-27:
I don't use PhoneGap build much. From what I recall as long as you add gap:plugin into config.xml, the plugin should get installed.

    <gap:plugin name="com.megster.cordova.ble" version="0.1.0" />

This plugin doesn't have Chat example, which code are you using? I'd check config.xml for syntax errors and try uploading it again.

JohnHigbie, 2015-01-28:
Hi Don.

The example is from the plugin repository examples folder:
https://github.com/don/cordova-plugin-ble-central/tree/d43c7423e8b4eabb1d0fc801be2449b2a9182e88/examples/redbearlab.

I added the <gap:plugin... line to the example's config.xml and put it 
in the www folder for upload, which is how I've been doing it with 
bluetoothserial. I also tried including the ble.js, and adding its path 
in index.html. I tried having ble.js in both the js folder and the top 
www folder.

Do I need to be using the plugins.xml, or is that just for building it 
locally?

When I say that PGB doesn't show the plugin I'm referring to the plugins 
tab below the update & rebuild buttons. Also, the app doesn't do 
anything when you click "Refresh."

Best,
John

don, 2015-01-29:
Make sure your config.xml is [valid](http://validator.w3.org/#validate_by_input). I needed to add 

    xmlns:gap="http://phonegap.com/ns/1.0"

Once the `gap` namespace was added PhoneGap Build showed the plugin in the web UI.

Here's my test project https://github.com/don/pgb-test

The plugin installer adds ble.js and the other files into your project. You don't need to add the javascript or a reference in index.html. The plugins.xml file tells Cordova (or plugman) how to install the plugin. The tool reads that file, you can safely ignore it.

JohnHigbie, 2015-01-29:
Hi Don,
I guess that was the issue. I now have your pgb-test app functioning 
here, but only on IOS.
On a Nexus 7 tablet, the Biscuit (Red Bear BLE mini) doesn't appear in 
the app. The Refresh button doesn't bring up anything.
Also, I tried, but wasn't able to pair with the BLE mini in Settings. 
Using Android 4.4.2.
Any ideas?
Thanks,
John

don, 2015-02-11:
PhoneGap Build issue fixed. The demo and plugin works with Android too. Discussion moved to https://github.com/don/pgb-test/issues/1

closed 2015-02-11 by don