don/cordova-plugin-ble-central #36: ipad unrecognized selector sent to instance issue

shraddha1112, 2015-02-19:
Hello,
I am doing a project where we are connecting to our custom device and trying to communicate over BLE. We have done succesful round of implementation and test using your plugin with Android.
Now I have to test same with ios. 
I build and installed same application for ios. Now it is giving me error for scan itself.
     console.log("Testing plugin");
           $scope.status = "Searching...";
           ble.scan([], 5, function(devices) 
           {
             //console.log("FOunddddddddddd "+JSON.stringify(devices));
                            $scope.devices_.push(devices);
                            console.log("Address "+devices.name);
                            $scope.$apply();
            }, failure);

THe error is 

-[CBUUID UUIDString]: unrecognized selector sent to instance 0x176cf160

2015-02-19 11:40:53.930 ZenRelaxie[219:60b] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[CBUUID UUIDString]: unrecognized selector sent to instance 0x176cf160'

*** First throw call stack:

(0x30bd8f4b 0x3b36f6af 0x30bdc8e7 0x30bdb1cb 0x30b2a4d8 0x9efd7 0x9eb21 0x9b7ff 0x308ff8eb 0x308ffda7 0x30907ba3 0x3b852d1b 0x3b852d07 0x3b85978d 0x30ba3819 0x30ba20ed 0x30b0cc27 0x30b0ca0b 0x35800283 0x333b0049 0x9672b 0x3b877ab7)

libc++abi.dylib: terminating with uncaught exception of type NSException

At location

    // Create a new list of Service UUIDs as Strings instead of CBUUIDs
    NSMutableArray *serviceUUIDs = [dict objectForKey:CBAdvertisementDataServiceUUIDsKey];
    NSMutableArray *serviceUUIDStrings;
    if (serviceUUIDs) {
        serviceUUIDStrings = [[NSMutableArray alloc] initWithCapacity:serviceUUIDs.count];

        for (CBUUID *uuid in serviceUUIDs) {
            [serviceUUIDStrings addObject:[uuid UUIDString]]; //HEre is the breakpoint
        }

        // replace the UUID list with list of strings
        [dict removeObjectForKey:CBAdvertisementDataServiceUUIDsKey];
        [dict setObject:serviceUUIDStrings forKey:CBAdvertisementDataServiceUUIDsKey];

    }

Can you pelase help me?
I am a newbie to ios. So if i am missing any step please tell me.

Thanks,
Regards
 Shraddha

don, 2015-02-20:
What version of iOS are you running on? What version of XCode? What version of Cordova?

It looks like there's a problem parsing the advertising data. Do you know if there is a particular device that's breaking the code or any device?

In the debugger is `uuid` or `[uuid UUIDString]` nil?

If you don't care about advertising info you can comment out the `if (serviceUUIDs) {` block and test the other functions.

shraddha1112, 2015-02-23:
Hello,
sw_vers -productVersion : version 10.8.5
xcodebuild -version : 5.1
cordova -version : 4.2.0

As per your suggestion I directly tried to give static address to test other commands.
If i directly try to give static id for connect then it is working on android but not in ios.

var deviceID = "xx:xx:xx:xx:xx:xx";
  ble.connect(deviceID ,connectSuccess,failure);

or if i give something like BD922605-1B07-4D55-8D09-B66653E51BBA then also it is failing.
Error is could not find peripheral : xxxx
Same device is found in another utility and i am able to connect with it from that utility.

Please help.

Regards,
Shraddha

don, 2015-02-23:
The MAC address will work on Android, but on iOS you need the UUID that iOS discovers. iOS generates a random UUID and hides the real hardware information from you.

don, 2015-03-22:
Assuming this was not using the UUID from the scan results on iOS. If not, re-open the issue.

closed 2015-03-22 by don