don/cordova-plugin-ble-central #37: isConnected, how does it work?

bratanon, 2015-02-22:
When I read the documentation about *isConnected*, I got the feeling that the command is used to check if there is a established connection between the device running the code (my phone, android) and the other device (arduino with BLE shield) that you have to provide a "device_id" for.

I can connect to my arduino just fine, and by looking in the logs I can see that there is a connection established, BUT when I call  *isConnected* with the MAC address, it is returning a string `JSON error`.

Looking at the *ble.js*, it seems like the *device_id* isn't even used at all in the exec method.

I think this results in that the `String macAddress = args.getString(0);` in *BLECentralPlugin.java* is just an empty array (`[]`). 

Is this by design, or a bug, or do I miss something here?

don, 2015-02-23:
This is a bug. It was fixed by a pull request #35. If you install the plugin from github, you'll get the fix. 

This will be included in the next release.

bratanon, 2015-02-23:
Thanks, I consider this issue closed then.

closed 2015-02-23 by bratanon