don/cordova-plugin-ble-central #38: Passing a number for the characteristic to startNotification crashes in iOS

adamtegen, 2015-02-26:
This crashes in iOS: ble.startNotification(deviceId, 2220, 2221, success, failure);
This succeeds in iOS: ble.startNotification(deviceId, '2220', '2221', success, failure);

don, 2015-02-28:
UUIDs should be passed as strings rather than ints

don, 2015-03-01:
I've been thinking about trying to convert ints to strings, but I think it's a bad idea. For example you entered 2220 as an int which is 0x8AC hex, so you code would still fail. You'd need to pass in 0x2220 (or 8736) for this to work, but I think that adds a lot more confusion. If you have suggestions on how to improve the documentation to indicate this, please let me know.

closed 2015-03-22 by don

adamtegen, 2015-08-25:
The added documentation is great.  Thanks.

commit d41b3881 (2015-03-22): UUIDs fixes #38