don/cordova-plugin-ble-central #39: Android issues: disconnect with Android 4.4.2 and multiple device communication

alesciack, 2015-02-27:
Hi there, 
I'm developing an app for controlling a BLE lamp. The plugin works fine with iOS (also allowing multiple devices being connected and controlled), but has some troubles with Android. In particular: 
- testing it with a Samsung S4 with Android 4.4.2, the app isn't able to safely disconnect from a peripheral (the app crashes when trying to disconnect); 
- testing it with a Sony Xperia L with Cyanogenmod12 unofficial (just because it was the only way to enable BLE on the Sony Xperia L), everything seems to work fine but writing to multiple connected device: as soon as the app tries to write to more than a single device, just one receives the message, and the others fall in a 'not-connected not-disconnected' state.
I'm trying to figure out which are the problems (maybe the Phonegap code itself is bugged), and I'm still in a preliminary stage of development, but just to be sure, are you aware of some kind of pitfalls that could explain this behavior?
Thanks for all the work you have done!
alessandro

don, 2015-02-28:
If you can provide simple projects that demonstrates the issues, it will be easier for me to track down any bugs.

alesciack, 2015-03-06:
While writing a simple code I could provide you, I was able to make it works (at least testing it with the devices I have). The only change I did is about writing to multiple devices, just make the following write operation wait until the current is completed. 
I'm going to try this code also in S4 to find out if everything is fine. Anyway, I will post the code, maybe it could be useful to be analyzed or as sample code.

don, 2015-03-22:
@alesciack did this work on your S4?

alesciack, 2015-03-23:
Yes, it seems that the last version of my testing code is working also on S4. 
I just finished writing the code, I post the index.html and the index.js here. 

HTML - index.html

```
<!DOCTYPE html>
<!--
    Copyright (c) 2012-2014 Adobe Systems Incorporated. All rights reserved.

    Licensed to the Apache Software Foundation (ASF) under one
    or more contributor license agreements.  See the NOTICE file
    distributed with this work for additional information
    regarding copyright ownership.  The ASF licenses this file
    to you under the Apache License, Version 2.0 (the
    "License"); you may not use this file except in compliance
    with the License.  You may obtain a copy of the License at

    http://www.apache.org/licenses/LICENSE-2.0

    Unless required by applicable law or agreed to in writing,
    software distributed under the License is distributed on an
    "AS IS" BASIS, WITHOUT WARRANTIES OR CONDITIONS OF ANY
     KIND, either express or implied.  See the License for the
    specific language governing permissions and limitations
    under the License.
-->
<html>
    <head>
        <meta charset="utf-8" />
        <meta name="format-detection" content="telephone=no" />
        <meta name="msapplication-tap-highlight" content="no" />
        <!-- WARNING: for iOS 7, remove the width=device-width and height=device-height attributes. See https://issues.apache.org/jira/browse/CB-4323 -->
        <meta name="viewport" content="user-scalable=no, initial-scale=1, maximum-scale=1, minimum-scale=1, width=device-width, height=device-height, target-densitydpi=device-dpi" />

        <link rel="stylesheet" href="css/index.css" />
        <link rel="stylesheet" href="css/css.css" />
        <link rel="stylesheet" href="css/jquery.mobile-1.4.0.min.css" />
        
        <title>BLE test app</title>
    </head>
    <body>
        <div id="devices"></div>
        <div id="refreshDiv">
            <button type="button" id="refresh">Reset list</button>
        </div>
        <div id="sendingForm">
            String to send: <input type="text" name="sendstring" id="stringInputId"><br>
             <button type="button" onclick="sendString()">Send</button>
        </div>
        <div id="performTorrent">
        	msec delay: <input type="text" name="delayTorrent" id="delayTorrentId"><br>
            <button type="button" onclick="sendTorrent()">MultipleSend</button>
        </div>
        <div id="message" style="width:90%; max-height:40%; overflow:auto;"></div>

        <script type="text/javascript" src="jquery-1.7.2.min.js"></script>
        <script type="text/javascript" src="cordova.js"></script>
        <script type="text/javascript" src="js/index.js"></script>
        
        <script type="text/javascript">
            app.initialize();
        </script>
    </body>
</html>

```
JAVASCRIPT - index.js

```
[... 394 lines cut ...]
                setTimeout(nextMultiTorrentWrite, delayTorrent);
            } else if(delayTorrent == 0){
				nextMultiTorrentWrite();
        	}
        }, function(error) {
        	app.display("Error "+error+ " - on multiwrite on address " +connectedDevicesAddresses[where]);
        });
	}
}

function nextMultiTorrentWrite(){
    wValue++;
    torrentString = [wValue];
    if(wValue < 256){
        sendTorrent();
    } else {  //resetting
        wValue = 0;
        torrentString = [wValue];
    }
}

```
HTML, javascript and css file of phonegap project here (to make them work, you have to put javascript in a folder called 'js', and stylesheets in a folder called 'css'):

https://dl.dropboxusercontent.com/u/26254761/testBLEwwwFiles.zip

closed 2015-10-23 by don