don/cordova-plugin-ble-central #40: No stopScan method

ziXet, 2015-03-18:
What if the user wants to stop the scanning? I think it's not implimented in the plugin.

don, 2015-03-18:
[Scan](https://github.com/don/cordova-plugin-ble-central#scan) takes a timeout and stops automatically.

Scan returns devices as they're found. If you want to stop scanning before the timeout, you can just ignore the results.

ziXet, 2015-03-18:
Ok. But what about the following scenario:
click on start scan button, while scanning, user closes the scan view and the click on start scan button.
Now the question is, what would happen if I call scan while it's already scanning?

ziXet, 2015-03-18:
Also, it would be beneficial to a handler for timed out scan. Because I have to somehow be notified that scan is over and let the user to re-scan in case of failing to find the sensor.

don, 2015-03-18:
I'm not sure what happens if you call scan while it's already scanning. If you get an error or bad behavior please create a new issue with details.

I agree that sometimes it's nice to know when the scan ends. For now, you can use setTimeout to have another function that tracks when the scan ends. I realize this might be a pain in your use case. I'm trying to keep the API small and make it simple to use for most cases. Feel free to propose API changes or enhancements or submit pull requests to help make this plugin better.

don, 2015-03-19:
@ziXet try `startScan` and `stopScan` added in 9f4334ae1c6097a06b51e1cde2303ffd68ce0bfd

replace

    ble.scan([someUUID], 5, app.onDiscoverDevice, app.onError);

with 

    ble.startScan([someUUID], app.onDiscoverDevice, app.onError);
    setTimeout(ble.stopScan,
            5000,
            function() { alert("Scan complete"); },
            function() { alert("stopScan failed"); }
        );

ziXet, 2015-03-19:
Thanks Don. I only have experience in Android programming and would be more than happy to contribute to this awesome plugin.

don, 2015-03-21:
Fixed in 9f4334ae1c6097a06b51e1cde2303ffd68ce0bfd

closed 2015-03-21 by don