don/cordova-plugin-ble-central #41: ReferenceError: ble is not defined

regaldisclaimer, 2015-03-18:
I didn't see a relevant previous issue, so I'm opening one.

I have an ionic project, and installed the plugin with the command

$ cordova plugin add com.megster.cordova.ble

The plugin seems correctly installed, but upon attempting to call ble.scan, I get:

ReferenceError: ble is not defined

Is there an installation step after the CLI installation that I'm missing? 

Thanks in advance

don, 2015-03-18:
That error sounds like the plugin is not installed correctly.

What versions of Ionic and Cordova are you using?
Are you seeing this on iOS or Android?

don, 2015-03-18:
I'm on OS X 10.10. Testing with Ionic 1.3.16 and Cordova 4.3.0. I created a project with `ionic start`, then added the plugin. `ble` is defined for Android after adding the plugin with cordova or with ionic. 

It looks like Ionic has changed since I last used it. I'm seeing a hook run after adding the plugin. `hooks/after_plugin_add/010_register_plugin.js`. Maybe this is the piece you're missing?

regaldisclaimer, 2015-03-19:
That's what I figured as well. 

I'm on the latest version of Ionic (1.3.16) developing for Android. I do have the hook that you mention in the folder, and I've already attempted re-installing the plugin, as well as re-adding android as a platform. Do you have any other idea as to what may be going wrong?

Thanks!

don, 2015-03-19:
Maybe it's something with your environment? Are you on OS X, Windows or Linux? What version of Node are you running. Any way you can share the code with me so I can take a look?

I use cordova plugin add since most of the time. I think iconic is just calling Cordova so it shouldn't matter which you use.

regaldisclaimer, 2015-03-19:
Perhaps. I'm using node 0.10.32 on Windows. I'll try upgrading node & npm and try to install with ionic on a new project and report back within 24 hours. Thanks so much!

regaldisclaimer, 2015-03-19:
Still no luck, unfortunately. I started an ionic's tab sample project to test the plugin via:

>ionic start someName tabs

and created a service for the bluetooth signal. I [copy-pasted the code for my services.js file](https://gist.github.com/regaldisclaimer/690b93648b1568d9039a). The last `.factory('bleService')...` is the service I added to test the plugin.  Basicallly, I'm just trying to get the plugin to work.

I still get 

[image]

any ideas?

Thank you!

don, 2015-03-22:
Here's a sample Ionic BLE project https://github.com/don/ionic-ble

Tested on OS X 10.10 and Windows 8.1 with Ionic 1.3.16

regaldisclaimer, 2015-03-22:
Thank you so much! I looked through the commits, and it doesn't look like I'm missing any key steps. I'll spin myself a unix environment, test things myself, and leave an update once I identify the problem.

Thanks for your help!

closed 2015-03-22 by regaldisclaimer

jried31, 2015-05-03:
Hello, also am having this issue. Not using Ionic, but Javascript/Jquery and HTML...This plugin is not being included in the scope for some reason. Any possible thoughts on how to force include this so it's in scope?

don, 2015-05-03:
@jried31 maybe you need to upgrade your version of Cordova or your generated project? Try creating a new test project and see if the plugin works there.

bond-thread, 2015-05-08:
Hi, I'm having the same issue ("Reference Error: Can't find variable: ble"; works fine on Android but not iOS). I've updated everything and checked all the other potential fixes here. Has anyone figured out a fix for this yet? I'm using Ionic.

jried31, 2015-05-10:
@don Your correct, I originally deployed the project based on the files in your example code. Doing so results in the scoping issues due to out of date library files. However, after creating the project from scratch with the most up-to-date libraries and adding the BLE library and pieces of your sample code, it worked.

@bond-thread At 1st, I was an avid user of Ionic due to it's AngularJS integration, however I ran into many scoping issues with AngularJS that it became more of a pain to deal with. I found success using a simpler (more traditional) method like the template here works https://github.com/app-o-mat/jqm-cordova-template-project