don/cordova-plugin-ble-central #42: Indicate issue

shraddha1112, 2015-03-20:
Hello,
I am developing an application with Ionic Cordova. We have our custom eval kit. It has 6 characteristics which i can read and write. But I cannot register to indicate for particular characteristic. I am not getting success callback it is giving invalid action. 
'IndicateUUID':'xxxx',
  ble.indicate( device.id, UUIDs.ServiceUUID, UUIDs.IndicateUUID, indicateSuccessCallBack, failure);

Please help.

Regards,
Shraddha

shraddha1112, 2015-03-20:
It seems indicate is not implemented in android? Is it so? 
I tried using startNotification for same but I cannot get notification as well. Not even for battery. 
Please help we are in phase of delivery for release. 

Regards,
Shraddha

don, 2015-03-20:
Indicate is not implemented on iOS or Android yet. I'm working on this, but it might take a bit.

startNotification works on both iOS and Android. 

Use something like [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) to verify notification is enabled for your battery service (notification is optional.) 

Are you able to register for notifications for the battery service but just not seeing changes?

shraddha1112, 2015-03-22:
Hello,
 I am getting battery service notification on anroid. But we want indicate. Will it take long time to do so? 

Regards,
Shraddha

don, 2015-03-22:
`startNotification` supports notification and indication in version 0.1.6

closed 2015-03-22 by don