don/cordova-plugin-ble-central #45: ios Write not working

shraddha1112, 2015-04-01:
Hello,
I am trying to test application which is running in Android same one on ios 8. Here connect is working and read is working. But write is not working. It is not giving error and on hardware side nothing is getting recieved. is there any difference for write command in android and ios?
  var configData = new Uint8Array(1);
                configData[0] = Params.IValue;
                ble.writeWithoutResponse(DeviceID, UUIDs.ServiceUUID, UUIDs.VUUID, configData.buffer, onParamWriteSuccess, failure);

I am not getting failure in ios. But getting success log. But on hardware side i checked from putty nothing is getting logged for write. Connect is working fine. And Read as well.

Regards,
Shraddha

shraddha1112, 2015-04-01:
Please help.

don, 2015-04-02:
Check the characteristic properties, maybe you need ble.write instead of ble.writeWithoutResponse?

Unfortunately with writeWithoutResponse the peripheral is free to ignore your write request.

Try using a tool like [LightBlue](https://itunes.apple.com/us/app/lightblue-bluetooth-low-energy/id557428110?mt=8) or [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) to explore the characteristics and send write requests.

shraddha1112, 2015-04-24:
ble.write is working but ble.writeWithoutResponse is not working in ios. We cannot go ahead with ble.write since it takes long time to respond. For Android ble.writeWithoutResponse is working. But not for ios.

don, 2015-04-24:
I suspect the problem is your peripheral does not support write without response.

After you connect, you'll get a peripheral object in the success callback. Log the JSON and post the details here. I'd like to see the properties for the characteristic that you want to write to.

shraddha1112, 2015-04-27:
I am able to do writeWithoutResponse with same peripheral with android App. But same code when I compile for ios and run writeWithoutResponse is not working. 
Callback object is null.
     var configData = new Uint8Array(1);
                configData[0] = Params.VIntValue.toString();
                ble.writeWithoutResponse(devID, UUIDs.ServiceUUID, UUIDs.VbnIntensityUUID, configData.buffer, onParamWriteSuccess, failure);
          

Thanks for your help.

don, 2015-05-08:
Can you post the peripheral information that is passed as JSON to the connect success callback.

I need to see the properties for the characteristic you're attempting to write. "write" and  "write without response" are two different properties. I think iOS is stricter about enforcing the properties, where as Android is more flexible and just tries to write the data.

shraddha1112, 2015-05-22:
Hello Don,
Sorry for late reply since was busy in release cycle.
So here is the response:
{"characteristics":[{"characteristic":"2a00","service":"1800","properties":["Read","Write"]},{"characteristic":"2a01","service":"1800","properties":["Read"]},{"characteristic":"2a04","service":"1800","properties":["Read"]},{"descriptors":[{"uuid":"2902"}],"characteristic":"2a19","service":"180f","properties":["Read","Notify"]},{"characteristic":"058e1555-f52f-8f88-870d-67ce119e315c","service":"058e0555-f52f-8f88-870d-67ce119e315c","properties":["Read","Write"]},{"characteristic":"058e2555-f52f-8f88-870d-67ce119e315c","service":"058e0555-f52f-8f88-870d-67ce119e315c","properties":["Read","Write"]},{"characteristic":"058e3555-f52f-8f88-870d-67ce119e315c","service":"058e0555-f52f-8f88-870d-67ce119e315c","properties":["Read","Write"]},{"characteristic":"058e4555-f52f-8f88-870d-67ce119e315c","service":"058e0555-f52f-8f88-870d-67ce119e315c","properties":["Read","Write"]},{"characteristic":"058e5555-f52f-8f88-870d-67ce119e315c","service":"058e0555-f52f-8f88-870d-67ce119e315c","properties":["Read","Write"]},{"characteristic":"058e6555-f52f-8f88-870d-67ce119e315c","service":"058e0555-f52f-8f88-870d-67ce119e315c","properties":["Read","Write"]},{"characteristic":"058e1055-f52f-8f88-870d-67ce119e315c","service":"058e0555-f52f-8f88-870d-67ce119e315c","properties":["Read","Write"]},{"characteristic":"058e1155-f52f-8f88-870d-67ce119e315c","service":"058e0555-f52f-8f88-870d-67ce119e315c","properties":["Read","Write"]},{"characteristic":"058e7555-f52f-8f88-870d-67ce119e315c","service":"058e0555-f52f-8f88-870d-67ce119e315c","properties":["Read","Write"]},{"descriptors":[{"uuid":"2902"}],"characteristic":"058e8555-f52f-8f88-870d-67ce119e315c","service":"058e0555-f52f-8f88-870d-67ce119e315c","properties":["Read","Notify"]},{"descriptors":[{"uuid":"2902"}],"characteristic":"058e9555-f52f-8f88-870d-67ce119e315c","service":"058e0555-f52f-8f88-870d-67ce119e315c","properties":["Read","Notify"]},{"characteristic":"2a29","service":"180a","properties":["Read"]},{"characteristic":"2a27","service":"180a","properties":["Read"]},{"characteristic":"2a26","service":"180a","properties":["Read"]}],"advertising":{},"id":"xx:xx:xx:xx:xx:xx","services":["1800","1801","180f","058e0555-f52f-8f88-870d-67ce119e315c","180a"],"rssi":-60,"name":"xxxxxxxx"}

We are writing for all the properties mentioned as write.

Regards,
Shraddha

shraddha1112, 2015-05-25:
Any help regarding this issue?

don, 2015-05-30:
Reformatted JavaScript

    {
        "characteristics": [{
            "characteristic": "2a00",
            "service": "1800",
            "properties": ["Read", "Write"]
        }, {
            "characteristic": "2a01",
            "service": "1800",
            "properties": ["Read"]
        }, {
            "characteristic": "2a04",
            "service": "1800",
            "properties": ["Read"]
        }, {
            "descriptors": [{
                "uuid": "2902"
            }],
            "characteristic": "2a19",
            "service": "180f",
            "properties": ["Read", "Notify"]
        }, {
            "characteristic": "058e1555-f52f-8f88-870d-67ce119e315c",
            "service": "058e0555-f52f-8f88-870d-67ce119e315c",
            "properties": ["Read", "Write"]
        }, {
            "characteristic": "058e2555-f52f-8f88-870d-67ce119e315c",
            "service": "058e0555-f52f-8f88-870d-67ce119e315c",
            "properties": ["Read", "Write"]
        }, {
            "characteristic": "058e3555-f52f-8f88-870d-67ce119e315c",
            "service": "058e0555-f52f-8f88-870d-67ce119e315c",
            "properties": ["Read", "Write"]
        }, {
            "characteristic": "058e4555-f52f-8f88-870d-67ce119e315c",
            "service": "058e0555-f52f-8f88-870d-67ce119e315c",
            "properties": ["Read", "Write"]
        }, {
            "characteristic": "058e5555-f52f-8f88-870d-67ce119e315c",
            "service": "058e0555-f52f-8f88-870d-67ce119e315c",
            "properties": ["Read", "Write"]
        }, {
            "characteristic": "058e6555-f52f-8f88-870d-67ce119e315c",
            "service": "058e0555-f52f-8f88-870d-67ce119e315c",
            "properties": ["Read", "Write"]
        }, {
            "characteristic": "058e1055-f52f-8f88-870d-67ce119e315c",
            "service": "058e0555-f52f-8f88-870d-67ce119e315c",
            "properties": ["Read", "Write"]
        }, {
            "characteristic": "058e1155-f52f-8f88-870d-67ce119e315c",
            "service": "058e0555-f52f-8f88-870d-67ce119e315c",
            "properties": ["Read", "Write"]
        }, {
            "characteristic": "058e7555-f52f-8f88-870d-67ce119e315c",
            "service": "058e0555-f52f-8f88-870d-67ce119e315c",
            "properties": ["Read", "Write"]
        }, {
            "descriptors": [{
                "uuid": "2902"
            }],
            "characteristic": "058e8555-f52f-8f88-870d-67ce119e315c",
            "service": "058e0555-f52f-8f88-870d-67ce119e315c",
            "properties": ["Read", "Notify"]
        }, {
            "descriptors": [{
                "uuid": "2902"
            }],
            "characteristic": "058e9555-f52f-8f88-870d-67ce119e315c",
            "service": "058e0555-f52f-8f88-870d-67ce119e315c",
            "properties": ["Read", "Notify"]
        }, {
            "characteristic": "2a29",
            "service": "180a",
            "properties": ["Read"]
        }, {
            "characteristic": "2a27",
            "service": "180a",
            "properties": ["Read"]
        }, {
            "characteristic": "2a26",
            "service": "180a",
            "properties": ["Read"]
        }],
        "advertising": {},
        "id": "xx:xx:xx:xx:xx:xx",
        "services": ["1800", "1801", "180f", "058e0555-f52f-8f88-870d-67ce119e315c", "180a"],
        "rssi": -60,
        "name": "xxxxxxxx"
    }

don, 2015-05-30:
Based on the peripheral info, all of the characteristics have the `Write` property. None of them have `WriteWithoutResponse` property.

It makes sense that iOS is failing for writeWithoutResponse. The behavior of this plugin is controlled by what CoreBluetooth does. 

So why does this work on Android? I'd guess the Android implementation is more lenient on what it accepts, so it'll writeWithoutResponse for a characteristic with the Write property.

closed 2015-09-11 by don