don/cordova-plugin-ble-central #46: Notify with ionic and ng-cordova

xam8re, 2015-04-14:
I've some problem with read and notify api. 
With android (i can't test on ios at this moment) i can attach to change notification but the array read is empty. Read function return always error. 

If I use your example https://github.com/don/cordova-plugin-ble-central/tree/master/examples/redbearlab it works. 

I'm started from this project: https://github.com/don/ionic-ble adding write and notify on the services.js. 
I can write but non receive.

xam8re, 2015-04-15:
Same using ngCordova module.

don, 2015-04-16:
What code are you running on your Ardunio? It's possible that the characteristic doesn't allow reads.

Can I see your code for handling the notification?

Since my [example](https://github.com/don/cordova-plugin-ble-central/tree/master/examples/redbearlab) works, I'm thinking the plugin is working OK with your hardware and something in your code needs to be adjusted.

closed 2015-09-17 by don