don/cordova-plugin-ble-central #47: Fix for devices that fail during characteristic notification

stakutis, 2015-04-14:
I've been using a BLE pedometer that has issues.  You ask it for its data, and it starts streaming notifications.  If there is a lot of data (more than a couple of seconds of transfer time), it just stops and dies.

In Android 5.1 there is new API, "gatt.requestConnectionPriority (1);", and I change the Peripheral.java/writeCharacteristic() method and call this just before it writes the characteristic, the device works perfectly well -- no data loss, doesn't stop, doesn't hang.  For some reason, this is essential, and I think a good fix (or, althernatively, add it as some sort of configuration setting).

don, 2015-04-16:
Do you call `gatt.requestConnectionPriority(BluetoothGatt.CONNECTION_PRIORITY_HIGH)` before every write request? Does it work if you set this one time after connecting?

stakutis, 2015-04-16:
I call it once, in "writeCharacteristic()", right before
gatt.writeCharacteristic().   So I guess it gets called each time I do a
"write" operation and maybe it that is unnecessary, but I haven't tried.  I
find the whole thing curious and not sure why it makes this problem go
away.  I mean, I could see it make it survive better in a noisy
environment, but in the fail-case, the device just goes completely off-line.

Chris Stakutis
978 764 3488
Twitter: @Stakutis

weisssean, 2016-05-27:
Is there any way to achieve this effect for older android versions?

peitschie, 2021-12-03:
This call has been exposed in https://github.com/don/cordova-plugin-ble-central/pull/714

See https://github.com/don/cordova-plugin-ble-central#requestconnectionpriority for usage instructions.

closed 2021-12-03 by peitschie