don/cordova-plugin-ble-central #48: Please remove tags 0.0.1 and 0.0.2

EddyVerbruggen, 2015-04-23:
Hi Don,

Sorry for the somewhat silly request, but can you remove [this](https://github.com/don/cordova-plugin-ble-central/releases/tag/0.0.1) and [this](https://github.com/don/cordova-plugin-ble-central/releases/tag/0.0.2) tag from your repo?

Or rename them to something like alpha-1 and alpha-2?

The problem we're having is that this plugin is liked to from Telerik AppBuilder which doesn't like the fact that in those two tagged versions the plugin.xml file is not in the root of the repo. You changed that since.

Thanks in advance!
Eddy

don, 2015-04-29:
@EddyVerbruggen I can adjust these tags now, but hopefully you can fix Telerik AppBuilder so this isn't a problem in the future with other tags in other repos. What are you matching on would foo-0.0.1 or 0.0.1-foo fix the issue?

EddyVerbruggen, 2015-04-29:
Hi Don, agreed, but at the moment this is the only issue we're having (I'm not personally in a position to fix those issues btw).

We list only tags which contain at least two component version numbers in them (e.g. 1.3, 0.3.2, etc.). So if you’d like to retain the tags please rename them to alpha-1, alpha-001 or something else which won’t have a version in it.

Thanks for the help!

closed 2015-05-03 by don