don/cordova-plugin-ble-central #49: peripheral/advertiser mode

titoasty, 2015-04-27:
now that server mode is stable on android, it would be a great enhancement to add it

don, 2015-04-29:
Peripheral mode exists for Android, but it's not necessarily stable or widely supported.

I think peripheral mode is different enough that it will be a different plugin rather than mixing with this one. I have some experimental code but nothing that can be published yet.

titoasty, 2015-04-29:
Yes you are totally right about making another plugin.
I'm really interested and in need of it, so if I can be useful on this, I can help you for android and probably windows phone versions.

apptum, 2015-07-31:
hey don, as I'm searching for a phonegap library to advertise/broadcast as a beacon (in background) - you are not by any chance working on something like that?

don, 2015-08-27:
@apptum other things have been distracting me from the peripheral plugin so I don't have anything publishable yet. For advertising as a beacon you might try https://github.com/petermetz/cordova-plugin-ibeacon, although it might be iOS only for what you need.

apptum, 2015-08-30:
@don thanks for the hint! Peters plugin is great, but it seems it's no longer in development and though it can advertise as beacon, it can't do so in background :( But hopefully he just pauses.

don, 2016-05-07:
http://github.com/don/cordova-plugin-ble-peripheral

closed 2016-05-07 by don

apptum, 2016-05-31:
@don no way! Hail the Don!