don/cordova-plugin-ble-central #51: StopNotification on Android

agumbs1, 2015-05-05:
Does stopNotification work on Android?  I called it and received an "invalid action."  I looked quickly at the source and it seems that this action is not present in the .java source.

Much appreciated.

don, 2015-05-08:
It looks like I never implemented this on Android. I'll get it in a future version.

Andy-Donegan, 2015-05-15:
I am also looking for the same stopNotification functionality, thank you for responding, I have also just found the issue with stopNotification. Thanks to agumbs1 post and Don's reply I can stop banging my head against this fluffy pillow.

don, 2015-06-06:
Need to implement Peripheral.removeNotifyCallback which is the inverse of Peripheral.registerNotifyCallback. Also need to add queuing of commands.

JohnMcLear, 2015-06-13:
+1 this caught me out today./

don, 2015-06-15:
@JohnMcLear Unfortunately it's not a trivial fix so I'm dragging my feet on this. If you're disconnecting, there's not need to stop notifications first. If you're staying connected, and you really need to stopNotifications, you can set a flag in javascript and just ignore any incoming data until this is fixed.

alexissantos, 2015-07-04:
+1 here as well.

I might be wrong, but it seems like subscriptions may be stacking with every reconnect. Say, for example, I connect a total of six times. Data I send once from my Arduino will be picked up by the mobile app six times. stopNotifications might help out here. If I stop the app from running -- not just hit home and navigate to a different app -- and open it up again, it looks like the notifications are reset.

This could also be an issue with Meteor, since I'm using the plugin with it instead of standalone Cordova.

Most importantly, however, thanks so much for building this @don! :) I'll work around this in the meantime.

don, 2015-07-06:
@alexissantos if you disconnect, the notifications should reset. On another project (https://github.com/don/cordova-plugin-rfduino/issues/4) I was using `gatt.disconnect` instead of `gatt.close` and saw similar issues. This plugin is using `gatt.close`. Maybe the API changed? Can you create a new bug? Details about your Android version and device would be helpful to try and reproduce.

alexissantos, 2015-07-07:
@don Sure, I'll create a new bug!

alexissantos, 2015-07-07:
@don Okay, I've pinned down the bug a bit more specifically. It happens when the Arduino is turned off and I connect the phone to it again. I've created an issue here: https://github.com/don/cordova-plugin-ble-central/issues/71

closed 2015-12-03 by don

alexissantos, 2015-12-03:
Thanks, Don! :)

commit b6fda752 (2015-12-03): stopNotification for Android fixes #51