don/cordova-plugin-ble-central #52: Advertising data is empty on Android + Fix

robrow, 2015-05-12:
iOS works like a charm but on Android 5 the "advertising" field is empty.
Fix in ble.js, line 40
```javascript
function convertToNativeJS(object) {
    Object.keys(object).forEach(function (key) {
        var value = massageMessageNativeToJs(object[key]);
        object[key] = value;
        if (typeof(value) === 'object') {
            convertToNativeJS(value);
        }
    });
}
```
Explanation

```javascript
function convertToNativeJS(object) {
    Object.keys(object).forEach(function (key) {
        var value = object[key];
        object[key] = massageMessageNativeToJs(value);
        if (typeof(value) === 'object') {
            convertToNativeJS(value);
        }
    });
}
```

1. var value = object[key];
2. object[key] = massageMessageNativeToJs(value);
3. if (typeof(value) === 'object')

The variable "value" gets never updated. So the "Base64 ArrayBuffer" is processed twice and is unusable after that.

luposlip, 2015-05-28:
+1 to have this fixed.
Also it would be nice to have a common format (as you ask for input to in the documentation). I suggest a map for both ios and android (and other supported platforms).

don, 2015-06-01:
I'm not seeing what you describe. On Nexus 5, Android 5.1.1, Cordova 5.0.0, BLE 0.1.7  I'm getting advertising info.

Can you try this code on your phone?

    var app = {
        initialize: function() {
            this.bindEvents();
        },
        bindEvents: function() {
            document.addEventListener('deviceready', this.onDeviceReady, false);
        },
        onDeviceReady: function() {
            ble.scan([], 5, app.onDiscoverDevice, app.onError);
        },
        onDiscoverDevice: function(device) {
            // log the peripheral info
            console.log(JSON.stringify(device));
            // log the advertising info
            var advertising = new Uint8Array(device.advertising);
            console.log(JSON.stringify(advertising));
        },
        onError: function(reason) {
            console.log("Error " + reason);
        }
    };

don, 2015-09-17:
Can not duplicate. I see advertising info on Android.

closed 2015-09-17 by don

iptelephony, 2016-03-20:
I can confirm that this happens on Nexus 7 Android 6.0.1 and RedMi 2 Android 4.4.4, using plugin version 1.0.1.

The issue lies in ble.js line 18:
```
var stringToArrayBuffer = function(str) {
    var ret = new Uint8Array(str.length);
    for (var i = 0; i < str.length; i++) {
        ret[i] = str.charCodeAt(i);
    }
    // TODO would it be better to return Uint8Array?
    return ret.buffer;
};

```
changing the last line to `return ret` solves the issue.

don, 2016-03-21:
@iptelephony Can you check again? stringToArrayBuffer returns an ArrayBuffer. If you log the ArrayBuffer, it looks empty {}, but it's not. You can verify this by checking `buffer.byteLength`. Or by creating a new Uint8Array from the ArrayBuffer with `new Uint8Array(buffer)`;