don/cordova-plugin-ble-central #53: Read the RSSI for a connected device

manouBayi, 2015-05-13:
Hello,
I'm looking for a way to constantly get RSSI value of a ble device to make some distance-rssi approximation.
 
Thanks

robrow, 2015-05-15:
Hi,
could you use the RSSI value stored in advertising packets?

don, 2015-05-28:
This is possible. I need to add a new API call the looks something like https://github.com/don/BluetoothSerial#readrssi

manouBayi, 2015-06-03:
ok , I'll try by adding 'BluetoothSerial '

 Envoyé le : Jeudi 28 mai 2015 7h21
 Objet : Re: [cordova-plugin-ble-central] Read the RSSI for a connected device (#53)

don, 2015-06-03:
@manouBayi I don't think BluetoothSerial will help you. A new gerRSSI function needs to be added  to this plugin.

manouBayi, 2015-06-03:
ok, thank you

 Envoyé le : Mercredi 3 juin 2015 14h03

Andy-Donegan, 2015-07-17:
Hi,
I am now looking for a way to update the RSSI as well.

Found this link on SO would this be the correct way to go about and implement within the Plugin ?

I have to be honest I would not have a clue where to put this code, but if I am on the correct track I will give it  a go.

http://stackoverflow.com/questions/18625819/continual-bluetooth-le-signal-strength-on-android

if(newState == BluetoothProfile.STATE_CONNECTED) 
{
   TimerTask task = new TimerTask()
   {
      @Override
      public void run()
      {
         mBluetoothGatt.readRemoteRssi();
      }
   };
   Timer rssiTimer = new Timer();
   rssiTimer.schedule(task, 1000, 1000);
}

mrAddy, 2016-01-20:
Do we have any idea on when this enhancement will be implemented, or any workaround to achieve it ?

don, 2016-01-20:
There's no work-around, it just needs to be implemented. I don't have a current timeline to implement. This would be nice to have, but I'm probably not going to add it until one of my projects needs it.

snowbreeze, 2016-02-01:
Can anyone please implement this

snowbreeze, 2016-02-01:
I have implemented it on android.. dont have ios and wp

don, 2017-08-22:
https://github.com/don/cordova-plugin-ble-central#readrssi

closed 2017-08-22 by don