don/cordova-plugin-ble-central #54: CoreBluetooth[API MISUSE] <CBCentralManager: 0x17d84ab0> can only accept commands while in the powered on state

itree, 2015-05-20:
Hi, 

This happens during ble.scan.  It works fine on android, but on iOS 7 I get this error 

CoreBluetooth[API MISUSE] <CBCentralManager: 0x17d84ab0> can only accept commands while in the powered on state

Is this because ble is not supported on iOS 7?

don, 2015-05-28:
The plugin originally worked with iOS 7, but I've only been using with iOS 8 lately.

Is Bluetooth enabled (or powered on) before you run the scan?

don, 2015-05-30:
I'm having trouble tracking down an iOS7 device to test this. I can duplicate the issue on an iPhone 4 running 7.1.2. 

BLE is not supported on iPhone 4, you need at least an iPhone 4s.

Do you see this in your logs when the plugin initializes?
2015-05-30 11:22:24.069 BluefruitLE[12292:60b] Status of CoreBluetooth central manager changed 2 State BLE unsupported (CBCentralManagerStateResetting)

@itree what hardware are you using?

itree, 2015-05-31:
Hey Don, thanks.  I think that is the problem. I don't have the proper hardware.  Thanks.

hervethouzard, 2017-09-26:
Hello,

Same problem for me with an iPhone SE and iOs 11:
[CoreBluetooth] API MISUSE: <CBCentralManager: 0x1c0672900> can only accept this command while in the powered on state

The scan does not return anything.

hervethouzard, 2017-09-27:
Hello,

The problem seems partially solved with the new iOs update from yesterday (iOs 11.0.1).
I did not changed anything in my app except that I updated my iDevices and now the app runs...

But there's still a problem, I can see, again, the BLE devices but some devices are repeated twice.
You can see the result here: https://imgur.com/VPVW9s1

I suppose it's a problem related to iOs.