don/cordova-plugin-ble-central #55: Writing more than 20 bytes to a Characteristic

FwiLL, 2015-05-22:
Hi,

I'm not able to write more than 20 bytes at a time to a Characteristic.

I know the MTU limit of 23 bytes, however, is there a way to write it fragmented into multiple packets ?

Thanks

don, 2015-06-01:
I don't have a good answer for you. I think iOS might handle large messages automatically. In the past I've broken large messages into multiple < 20byte messages.

don, 2015-09-18:
Writing more than 20 bytes is working now. Android and iOS are transparently handling the offsets for the blob, so from the plugin's perspective large writes work. Your peripheral and service will need to support the large characteristics. This probably doesn't work on many versions of Android.

I was able write up to 512 bytes on iOS 8.4.1, iOS 9.0 and Android 5.1.1 (Nexus 5) using Cordova 5.3.1 and the 1.0.0 (snapshot) of cordova-plugin-ble-central.

The peripheral was [test.js](https://github.com/sandeepmistry/bleno/blob/master/test.js) running on OS X 10.10.5 with Node.js v0.12.7. Bleno handles offsets to the handled in [onWriteRequest](https://github.com/sandeepmistry/bleno#write-requests).

	var success = function() {
	    alert('write success');
	};
	var failure = function(error) {
	    alert('write failed ' + error);
	};

	var data = new Uint8Array(100);
	for (var i = 0; i < data.length; i++) {
	    data[i] = 0xFF;
	}

	ble.write(connectedPeripheral.id,
	    'ffffffff-ffff-ffff-ffff-fffffffffff0',
	    'ffffffff-ffff-ffff-ffff-fffffffffff4',
	    data.buffer,
	    success,
	    failure
	);

closed 2015-09-18 by don

don, 2015-09-18:
@sandeepmistry clarified a few points for me

Writing 512 byte characteristics worked in my test because of CoreBluetooth. This wouldn't work as well on Linux. On Linux you should be able to do up to 100 bytes.

Write length is going to be hardware dependent for other devices. For example the nRF8001 can only write 20 bytes, but should be able to read up to 512.

airbly, 2016-05-03:
Hi,

I am having issues writing more than 20 bytes at a time on Android 4.4.2.  

It appears that you can direct Android to negotiate an MTU as it does not automatically do so.

http://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#requestMtu%28int%29
public boolean requestMtu (int mtu)

Is this something you can consider adding to this API?  ble.requestMTU(size) or something to that effect?

Thanks for all your help otherwise.  This library is very handy.

- Chris

AarohGala, 2018-06-13:
You can send more than 20 bytes of data without breaking into chunks and including a delay. Every characteristics you are trying to write has an MTU value assigned. It's number of bytes you can write in one time. During the connection MTU values are exchanged and you can write those many bytes at a time. You can increase the mtu value on the server side (Max 512 bytes) and send that much bytes in one go.

For Android, you might want to request mtu manually after connecting with the server using

**requestMtu(int mtu)**

This is return true or false based on the mtu value you send. It will give a callback to onMtuChanged where Android device and server negotiate the maximum possible MTU value.

**onMtuChanged (BluetoothGatt gatt, int mtu, int status)**

and you can set MTU value in this function and can send more than 20 bytes in one go.