don/cordova-plugin-ble-central #56: Can't make it work :(

jacqueslelezard, 2015-05-23:
Hi! I managed to install this plugin on Cordova 5.0, when I install the app on android the bluetooth permission is asked, but then nothing seems to happen when launching something like  
   ble.scan([], 5, function(device) {
        console.log(JSON.stringify(device));
    }, failure);
Any idea of what could be the problem or a way to debug?
(Android 4.4, Bluetooth 4.0)
Thanks !

Andy-Donegan, 2015-05-24:
Hi,
The very first time I played with this plugin I dived straight in and had little or very trying results. 
Can I recommend that you use one of the excellent demonstrations that is included with the install of the package and try the one which suits which ever device you are trying to connect to and then try to commence debugging from a platform which you know works.

Unfortunately your details supplied will not gather much help as people need to know what you are trying to connect to etc, the demos are a little daunting for beginners like myself, but with a little time and testing you will be connecting and find that the plugin works very well and it just requires patience and a lot of reading :)

jacqueslelezard, 2015-05-24:
Ok thanks for your answer, I'll try the demo projects in a separate project.

Andy-Donegan, 2015-05-24:
Ensure you find a project specific to the type of device that you are trying to connect to.
Once you have the example implementing it will be so much easier for you to come back here or even better try Stack Overflow which is an excellent resource for help with code issues and help.

This forum here is primarily for helping the authors by notifying of bugs and issues or requesting features or offering code solutions. 

Stack Overflow is a huge resource of like minded individuals who try to help out each other. If you have the demo project and can not get that to work, posting on stack overflow with the code sections will get you an answer within a few hours.

Good luck, I have been through the process it is daunting but now I have an app fully communicating with my specific device all thanks to this excellent plugin.

closed 2015-05-29 by don