don/cordova-plugin-ble-central #58: Just does nothing! Probably my fault

ipauland, 2015-05-27:
I'm new to phonegap/cordova and using phonegap build. Absolute newbie
I'm just trying to get some very basic code working and nothing seems to happen when I run my project on an iPhone 5. Neither success or failure.
What have I missed?
Thanks, Paul

<!DOCTYPE html>
<html>
  <head>
    <title>BLE Properties</title>

    <script type="text/javascript" charset="utf-8" src="cordova.js"></script>
    <script type="text/javascript" charset="utf-8" src="ble.js"></script>
    <script type="text/javascript" charset="utf-8">

    // Wait for device API libraries to load
    //
    document.addEventListener("deviceready", onDeviceReady, false);

    // device APIs are available
    //
    function onDeviceReady() {
        var element = document.getElementById('deviceProperties');

ble.scan([], 5, function(device) {
    element.innerHTML = 'scan: '+JSON.stringify(device);

    },
    function() {
     element.innerHTML = 'scan failed <br />';
    }
);                          
    }
    </script>
  </head>
  <body>
    <p id="deviceProperties">Loading ble info...</p>
  </body>
</html>

don, 2015-05-28:
I'd expect your JavaScript isn't compiling. I suggest getting one of the examples to run, then try modifying it. https://github.com/don/cordova-plugin-ble-central/tree/master/examples

ipauland, 2015-05-28:
If I install the app on the iPhone 5, absolutely nothing happens. If I install the same app on my iPad Air it finds my Apple TV but not my MacBook Pro!
I haven't tried an Android device. None of the Apple equipment seems to discover my Arduino with a HC-06. My terminal program on an Android tablet seems to discover the Arduino without issue.

 My principle interest was to create using Phonegap a controller app (IOS and Android) that could communicate with the Arduino. It's a nightmare.

don, 2015-05-28:
I think the  HC-06 is a Bluetooth Classic device. This plugin only works with Bluetooth Low Energy.  iOS devices only work with Bluetooth Low Energy. 

Try the HC-06 on Arduino and Android with http://github.com/don/BluetoothSerial

ipauland, 2015-05-28:
Thanks Don. I have contacted the hardware guys and they were using a different module, so I'll be checking that out and trying again. I really don't understand why my tiny test using an iPhone and iPad behaved differently. I'll try again when the HC-06 is replaced.

closed 2015-05-29 by don