don/cordova-plugin-ble-central #59: Uncaught exception from plugin

nexus1892, 2015-05-29:
Hi,  I recently started to use your plugin for a project I am working on.  I have been able to successfully connect to my BLE device, however, when I try and write data to the tx characteristic I get an unhandled exception detailed below.  I have verified that the service and tx characteristic UUID's are correct.  I am using a HTC One m9 running Android 5.0.2.  Any help would be much appreciated!

05-29 14:26:28.726: D/BLEPlugin(28869): action = write
05-29 14:26:28.726: D/Peripheral(28869): Queuing Command com.megster.cordova.ble.central.BLECommand@2ce69ff5
05-29 14:26:28.726: D/Peripheral(28869): Processing Commands
05-29 14:26:28.726: D/Peripheral(28869): Write 12f10002-61fb-6947-22f7-8f42121bd123
05-29 14:26:28.726: D/BluetoothDevice(28869): mAddress: ED:2F:62:68:A6:B5
05-29 14:26:28.726: D/BluetoothDevice(28869): mAddress: ED:2F:62:68:A6:B5
05-29 14:26:28.726: D/BluetoothDevice(28869): mAddress: ED:2F:62:68:A6:B5
05-29 14:26:28.726: D/BluetoothDevice(28869): mAddress: ED:2F:62:68:A6:B5
05-29 14:26:28.726: D/BluetoothDevice(28869): mAddress: ED:2F:62:68:A6:B5
05-29 14:26:28.726: E/PluginManager(28869): Uncaught exception from plugin
05-29 14:26:28.726: E/PluginManager(28869): java.lang.NullPointerException: Attempt to invoke virtual method 'android.bluetooth.BluetoothGattCharacteristic android.bluetooth.BluetoothGattService.getCharacteristic(java.util.UUID)' on a null object reference
05-29 14:26:28.726: E/PluginManager(28869): 	at com.megster.cordova.ble.central.Peripheral.writeCharacteristic(Peripheral.java:366)
05-29 14:26:28.726: E/PluginManager(28869): 	at com.megster.cordova.ble.central.Peripheral.processCommands(Peripheral.java:440)
05-29 14:26:28.726: E/PluginManager(28869): 	at com.megster.cordova.ble.central.Peripheral.queueCommand(Peripheral.java:414)
05-29 14:26:28.726: E/PluginManager(28869): 	at com.megster.cordova.ble.central.Peripheral.queueWrite(Peripheral.java:396)
05-29 14:26:28.726: E/PluginManager(28869): 	at com.megster.cordova.ble.central.BLECentralPlugin.write(BLECentralPlugin.java:253)
05-29 14:26:28.726: E/PluginManager(28869): 	at com.megster.cordova.ble.central.BLECentralPlugin.execute(BLECentralPlugin.java:129)
05-29 14:26:28.726: E/PluginManager(28869): 	at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:114)
05-29 14:26:28.726: E/PluginManager(28869): 	at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:95)
05-29 14:26:28.726: E/PluginManager(28869): 	at org.apache.cordova.PluginManager.exec(PluginManager.java:130)
05-29 14:26:28.726: E/PluginManager(28869): 	at org.apache.cordova.CordovaBridge.jsExec(CordovaBridge.java:59)
05-29 14:26:28.726: E/PluginManager(28869): 	at org.apache.cordova.engine.SystemExposedJsApi.exec(SystemExposedJsApi.java:41)
05-29 14:26:28.726: E/PluginManager(28869): 	at com.android.org.chromium.base.SystemMessageHandler.nativeDoRunLoopOnce(Native Method)
05-29 14:26:28.726: E/PluginManager(28869): 	at com.android.org.chromium.base.SystemMessageHandler.handleMessage(SystemMessageHandler.java:28)
05-29 14:26:28.726: E/PluginManager(28869): 	at android.os.Handler.dispatchMessage(Handler.java:102)
05-29 14:26:28.726: E/PluginManager(28869): 	at android.os.Looper.loop(Looper.java:155)
05-29 14:26:28.726: E/PluginManager(28869): 	at android.os.HandlerThread.run(HandlerThread.java:61)

don, 2015-05-29:
In the success callback for [connect](https://github.com/don/cordova-plugin-ble-central#connect) you'll get a [peripheral object](https://github.com/don/cordova-plugin-ble-central#peripheral-data) back.

Can you `console.log(JSON.stringify(peripheral, null, 2))` and post the JSON here?

nexus1892, 2015-05-29:
05-29 16:47:49.386: D/SystemWebChromeClient(2732):   "name": "UNLABELED",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):   "id": "ED:2F:62:68:A6:B5",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):   "advertising": {},
05-29 16:47:49.386: D/SystemWebChromeClient(2732):   "rssi": -63,
05-29 16:47:49.386: D/SystemWebChromeClient(2732):   "services": [
05-29 16:47:49.386: D/SystemWebChromeClient(2732):     "1800",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):     "1801",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):     "180d",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):     "180f",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):     "180a"
05-29 16:47:49.386: D/SystemWebChromeClient(2732):   ],
05-29 16:47:49.386: D/SystemWebChromeClient(2732):   "characteristics": [
05-29 16:47:49.386: D/SystemWebChromeClient(2732):     {
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "service": "1800",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "characteristic": "2a00",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "properties": [
05-29 16:47:49.386: D/SystemWebChromeClient(2732):         "Read",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):         "Write"
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       ]
05-29 16:47:49.386: D/SystemWebChromeClient(2732):     },
05-29 16:47:49.386: D/SystemWebChromeClient(2732):     {
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "service": "1800",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "characteristic": "2a01",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "properties": [
05-29 16:47:49.386: D/SystemWebChromeClient(2732):         "Read"
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       ]
05-29 16:47:49.386: D/SystemWebChromeClient(2732):     },
05-29 16:47:49.386: D/SystemWebChromeClient(2732):     {
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "service": "1800",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "characteristic": "2a04",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "properties": [
05-29 16:47:49.386: D/SystemWebChromeClient(2732):         "Read"
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       ]
05-29 16:47:49.386: D/SystemWebChromeClient(2732):     },
05-29 16:47:49.386: D/SystemWebChromeClient(2732):     {
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "service": "180d",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "characteristic": "2a37",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "properties": [
05-29 16:47:49.386: D/SystemWebChromeClient(2732):         "Notify"
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       ],
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "descriptors": [
05-29 16:47:49.386: D/SystemWebChromeClient(2732):         {
05-29 16:47:49.386: D/SystemWebChromeClient(2732):           "uuid": "2902"
05-29 16:47:49.386: D/SystemWebChromeClient(2732):         }
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       ]
05-29 16:47:49.386: D/SystemWebChromeClient(2732):     },
05-29 16:47:49.386: D/SystemWebChromeClient(2732):     {
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "service": "180d",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "characteristic": "2a38",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "properties": [
05-29 16:47:49.386: D/SystemWebChromeClient(2732):         "Read"
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       ]
05-29 16:47:49.386: D/SystemWebChromeClient(2732):     },
05-29 16:47:49.386: D/SystemWebChromeClient(2732):     {
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "service": "180f",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "characteristic": "2a19",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "properties": [
05-29 16:47:49.386: D/SystemWebChromeClient(2732):         "Read",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):         "Notify"
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       ],
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "descriptors": [
05-29 16:47:49.386: D/SystemWebChromeClient(2732):         {
05-29 16:47:49.386: D/SystemWebChromeClient(2732):           "uuid": "2902"
05-29 16:47:49.386: D/SystemWebChromeClient(2732):         }
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       ]
05-29 16:47:49.386: D/SystemWebChromeClient(2732):     },
05-29 16:47:49.386: D/SystemWebChromeClient(2732):     {
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "service": "180a",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "characteristic": "2a29",
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       "properties": [
05-29 16:47:49.386: D/SystemWebChromeClient(2732):         "Read"
05-29 16:47:49.386: D/SystemWebChromeClient(2732):       ]
05-29 16:47:49.386: D/SystemWebChromeClient(2732):     }
05-29 16:47:49.386: D/SystemWebChromeClient(2732):   ]
05-29 16:47:49.386: D/SystemWebChromeClient(2732): }

don, 2015-05-30:
What service and characteristic are you trying to write to? Based on the debug info above, only device name 0x2a00 is writeable for generic access 0x1800. The heart rate monitor 0x180d and battery service 0x180f only have characteristics you can read and notify.

nexus1892, 2015-06-01:
I am trying to write to the tx characteristic 12f10002-61fb-6947-22f7-8f42121bd123 and receive on the rx characteristic 12f10003-61fb-6947-22f7-8f42121bd123.  The service UUID is 12f10001-61fb-6947-22f7-8f42121bd123.

nexus1892, 2015-06-01:
I am using the write method as follows:

ble.write("ED:2F:62:68:A6:B5", "12f10001-61fb-6947-22f7-8f42121bd123", "12f10002-61fb-6947-22f7-8f42121bd123", stringToBytes(data), function(success) {
															     console.log("Success!");
															 },
															 function(failure) {
																 console.log("Fail");
															 });

Let me know what you see wrong with this request.  Thanks!

don, 2015-06-01:
Based on the JSON you sent, I don't see the 12f10001-61fb-6947-22f7-8f42121bd123 service or the other characteristics being discovered on your device.

Can you connect to that peripheral with [nRF Master Control Panel](https://www.nordicsemi.com/eng/Products/nRFready-Demo-Apps/nRF-Master-Control-Panel-application) and see if the service is listed?

nexus1892, 2015-06-02:
You were correct.  There was a problem with the application loaded into the development board and it wasn't broadcasting the correct service.  That has now been fixed and I can successfully write to the tx characteristic.  Thanks for your help!

closed 2015-06-06 by don