don/cordova-plugin-ble-central #61: Still having same issue as Issue 31 - ble is not defined

travisryan, 2015-06-07:
I am still getting "ble is not defined" like the person in Issue 31. It seems they never proved they got it working either before it was closed. 

I took your steps in creating the new issue31 app and making the change you wanted. I still get the following...

D/CordovaLog( 6001): file:///data/data/com.adobe.phonegap.app/files/files/phonegapdevapp/www/index.html: Line 262 : Received Event: deviceready
I/chromium( 6001): [INFO:CONSOLE(262)] "Received Event: deviceready", source: file:///data/data/com.adobe.phonegap.app/files/files/phonegapdevapp/www/index.html (262)

D/CordovaLog( 6001): file:///data/data/com.adobe.phonegap.app/files/files/phonegapdevapp/www/js/index.js: Line 38 : Uncaught ReferenceError: ble is not defined
I/chromium( 6001): [INFO:CONSOLE(38)] "Uncaught ReferenceError: ble is not defined", source: file:///data/data/com.adobe.phonegap.app/files/files/phonegapdevapp/www/js/index.js (38)

don, 2015-06-10:
Did you follow the steps in #31 to create a new project?

You log messages look like you are using [PhoneGap Developer App](http://app.phonegap.com). If so that won't works since the PhoneGap Developer App doesn't come with this plugin.

travisryan, 2015-06-10:
I think my problem is I didn't realize that the 3rd party plugins won't 
work with the PhoneGap developer app connection. I had to actually do 
the install/run command and push everything to the phone over usb to get 
it to work.

don, 2015-06-10:
Yes, unfortunately with the developer app you're limited to the plugins they ship with .

closed 2015-06-10 by don