don/cordova-plugin-ble-central #64: .write returning integer instead of error?

JohnMcLear, 2015-06-13:
``ble.write("80:EA:CA:00:00:01", "ffa0", "ffa1", stringToBytes("1"), function(s){console.log("success:", s)}, function(f,e){console.log("failure:", f,e)});``

returns ``failure: 6``

Am I supposed to get a more explicit error message?

JohnMcLear, 2015-06-13:
``writeCommand`` does work, so I guess the write is just not getting a reply and somehow we need to translate the integer 6 into "no response received" or so?

don, 2015-06-16:
@JohnMcLear you need to pass an ArrayBuffer to write. Assuming you wanted to pass 17 to the service:

    var data = new Uint8Array(1);
    data[0] = 17;
    ble.write(deviceId, service, characteristic, data.buffer, success, failure);

JohnMcLear, 2015-06-16:
Okies, any chance we can have the error message let me know that please? :)

curcuz, 2015-07-16:
Hi there. i'm experiencing the same issue but i'm using the stringToBytes function provided in the readme. I noticed that i get the error only if the value i try to write is more than 20 characters. Any idea?

don, 2015-07-17:
Generally BLE is limited to 20 bytes. You can send more data but the server needs to handle the offset properly. Theoretically iOS and Android should chunk this data, but I haven't tested it.

don, 2015-09-18:
@JohnMcLear for Android, 6 is [GATT_REQUEST_NOT_SUPPORTED](https://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#GATT_REQUEST_NOT_SUPPORTED), probably from trying to write to a characteristic that only supports writeWithoutResponse.

A translation of status to String could be added here
https://github.com/don/cordova-plugin-ble-central/blob/c8a33a374508c5820d2147f60841e7e19077c7f4/src/android/Peripheral.java#L250