don/cordova-plugin-ble-central #66: ble.scan doesn't invoke callback

fend25, 2015-06-15:
I have Sony Xperia ZL (C6503), android 5.0, cordova@5.1.1, ionic@1.5.0

With simplest ionic app (even without angular at all) I launch this code:
```js
ionic.Platform.ready(function () {
  var device = ionic.Platform.device();
  console.log("Hey, I'm an " + device.platform);

  ble.enable(function () {
    console.log("Bluetooth is enabled");
    ble.scan([], 5, function (device) {
      alert(1);
      console.log('scanned');
      console.log(JSON.stringify(device));
    }, function (err) {
      alert(2);
      console.log('error');
      console.log(err);
    });
  }, function () {
    console.log("The user did *not* enable Bluetooth");
  });
});
```
And in console I have:
```
0     810077   log      Hey, I'm an Android
1     815711   log      Bluetooth is enabled
```
That's all. No any messages in console, nor alerts I get. I tried separately `startScan` and `stopScan`. `stopScan`'s callbacks are invoked, but `startScan` is absolutely silent.

don, 2015-06-15:
@fend25 that code looks ok (I didn't run it.)

Are you sure there's a BLE device to be discovered? The success callback is called when a BLE device is discovered. It'll be called 10x if there are 10 devices. If there no BLE devices discovered the success callback never gets called.

Use [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) to verify your phone can see some devices.

Try [bleno](https://github.com/sandeepmistry/bleno) if you need to create devices for development on Linux or OS X. The [echo service](https://github.com/sandeepmistry/bleno/tree/master/examples/echo) is a good place to start.

fend25, 2015-06-15:
Thank you for the answer and for the link to nRF app. It helped me localize problem.
Your library works well.

The device is xiaomi mi band.
The cause of the problem was that there is two apps work with mi band - official and unofficial. The unofficial one checks notification on android and sends commands to the band. As I can see, unofficial app blocked mi band's visibility, and binded mi band to itself.

closed 2015-06-15 by fend25