don/cordova-plugin-ble-central #67: startNotification issue

bardu, 2015-06-16:
I'm using Cordova 4.0.2 (which I think is the latest version) and a Samsung S6 with Android 5.0.2.

I can read from a characteristic, however, when trying to register notification neither the success nor the failure callback gets invoked.

In logcat there is a line:

06-15 17:18:49.043  27454-27510/ca.mobitec.ble D/BluetoothGatt﹕ setCharacteristicNotification() - uuid: 0000fff1-0000-1000-8000-00805f9b34fb enable: true

but afterward nothing happens.

Any ideas?

bardu, 2015-06-16:
Just got notice from device manufacture that notification isn't supported.

closed 2015-06-16 by bardu

don, 2015-06-16:
@bardu In the future if you have a device without documentation, you can use [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) to inspect the service and see what properties a characteristic supports.

bardu, 2015-06-16:
@don Yes, I have this app, however, for some reasons it could not connect to the device, but my test app and an app from the manufacture.