don/cordova-plugin-ble-central #68: ble.read, ble.startNotification, ble.write all fail

objectkuan, 2015-06-16:
I modify the Chats Controller in ionic-ble.

Now my app can scan and connect to my device, but the read, startNotification, and the write routines all end at failing callbacks.

Cannot figure out the reason.

      .controller('ChatsCtrl', function($scope, Chats, $cordovaDialogs, BLE) {
        $scope.chats = Chats.all();
        $scope.remove = function(chat) {
          // Chats.remove(chat);

          // keep a reference since devices will be added
          $scope.devices = BLE.devices;

          var success = function () {
              if ($scope.devices.length < 1) {
                  // a better solution would be to update a status message rather than an alert
                  alert("Didn't find any Bluetooth Low Energy devices.");
              }
              for (var i in $scope.devices) {
                var dev = $scope.devices[i];
                if (dev.name == "Happy-dev") {
                  console.log(dev);
                  BLE.connect(dev.id).then(
                      function(peripheral) {

                        var parseData = function(data) {
                          data = new Uint8Array(data);
                          var ret = '';
                          for (var i = 0; i < 20; i++) {
                            ret = ret + data[i] + ',';
                          }
                          return ret;
                        };

                        var characteristics = peripheral.characteristics;
                        console.log(characteristics);

                        
                        var SERVICEUUID = characteristics[0].service;
                        var DATAUUID    = characteristics[0].characteristic;
                        var ATCMDUUID   = characteristics[1].characteristic;
                        
                        // Write
                        var data = new Uint8Array(20);
                        data[0] = 0xaa;
                        data[1] = 0x55;
                        data[2] = 0x00;
                        data[3] = 0x00;
                        data[4] = 0x00;
                        data[5] = 0x00;
                        data[6] = 0x00;
                        data[7] = 0x00;
                        data[8] = 0x00;
                        data[9] = 0x00;
                        data[10] = 0x00;
                        data[11] = 0x00;
                        data[12] = 0x00;
                        data[13] = 0x01;
                        data[14] = 0x00;
                        data[15] = 0x00;
                        data[16] = 0x00;
                        data[17] = 0x00;
                        data[18] = 0x00;
                        data[19] = 0x00;
                        

                        // read
                        ble.read(dev.id, SERVICEUUID, DATAUUID, 
                          function(data) {
                            $cordovaDialogs.alert('read:' + parseData(data));
                          }, 
                          function() {
                            $cordovaDialogs.alert('BLE fails to read');
                          }
                        );
                        
                        // register notification
                        ble.startNotification(dev.id, SERVICEUUID, DATAUUID, 
                          function(data) {
                            $cordovaDialogs.alert('0:' + parseData(data));
                          }, 
                          function() {
                            $cordovaDialogs.alert('BLE fails to be notified');
                          }
                        );

                        ble.write(dev.id, SERVICEUUID, DATAUUID, data.buffer,
                          function() {
                            $cordovaDialogs.alert('write success 2');
                          },
                          function() {
                            $cordovaDialogs.alert('BLE fails to write 2');
                          }
                        );

                      }
                  );
                }
              }
          };

          var failure = function (error) {
              alert(error);
          };

          // initial scan
          BLE.scan().then(success, failure);
        }
      })

don, 2015-06-16:
You're modifying https://github.com/don/ionic-ble?

What's the error message when your code fails? 

Print the reason code passed to the failure callback. Get javascript errors using chrome://inspect or safari debugging. Look for native errors using `adb logcat` for Android or Xcode for iOS.

objectkuan, 2015-06-16:
Yes, I'm building a demo app from https://github.com/don/ionic-ble since it has nice scanning and connecting. But the my step is to send commands to a connected device. 

I tried again and updated my codes. It seems that something goes wrong when invoking *write*, it ends up in the failure callback and writes "BLE fails to write " to console.

Here is my safari debugging console information:

      [Log] SERVICEUUID: FAF0 (controllers.js, line 40)
      [Log] DATAUUID: FAF1 (controllers.js, line 41)
      [Log] dev.id: F71F5D50-86A7-D194-2C7D-9CF94E589D24 (controllers.js, line 42)
      [Log] 0:0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0, (controllers.js, line 80)
      [Log] read:0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0, (controllers.js, line 70)
      [Log] BLE fails to write (controllers.js, line 92)

don, 2015-06-16:
"BLE fails to write" is your error message. Try getting the error message passed to the failure callback

    ble.write(dev.id, SERVICEUUID, DATAUUID, data.buffer,
      function() {
        $cordovaDialogs.alert('write success 2');
      },
      function(reason) {
        $cordovaDialogs.alert('BLE fails to write 2 ' + reason);
      }
    );

Based on your log messages it looks like you're seeing a success call for notification and read but write is failing. Are you sure that service FAF0 characteristic FAF1 is writeable? Check the characteristic properties.

objectkuan, 2015-06-17:
Cool, I've ignored the reason parameter and it really helps.

It's quite interesting that I my *write* fails in the callback with an "unknown error" according to the reason. After I curiously change *write* to *writeWithoutResponse*, everything works as expected. Ah I think that's strange.

don, 2015-06-17:
That makes sense. The device probably only supports _writeWithoutResponse_ and not _write_. Look at the [peripheral object](https://github.com/don/cordova-plugin-ble-central#peripheral-data) that gets passed to connect success function. It contains a JSON description of the peripheral, services, and characteristics. The properties on a characteristic control what you can and can't do.

closed 2015-06-17 by don

levinmacua, 2018-12-25:
Hi @don 

The write callback was successful but they not write a value.

    this.ble.write(deviceId, serviceUUID,characteristicUUID,data.buffer).then(
      (success) => this.showAlert('','Command was successfully set. '),
      error => this.showAlert('Unexpected Error','Error updating power switch ' + error)
    ).catch(err=>{
      let errorAlert = this.alertCtrl.create({
        message: 'Error ' + err, 
        buttons:[
          {
            text: 'Ok',
            handler: () =>{
              
            }
          }
        ]
      });