don/cordova-plugin-ble-central #69: Reading Major and Minor

matamicen, 2015-06-24:
Hi, first of all thanks for share this solution.

I can't read or see the major and minor, I checked the Battery Example but I don't get it.

Can someone help me with some code to do that?

Thanks!

lok0613, 2015-06-29:
Me two, I even can't get the UUID in android

don, 2015-06-29:
@matamicen when you say Major and Minor are you referring to iBeacons? The battery example doesn't have major and minor ids.

@lok0613 Android give you the MAC address for devices. iOS generates a UUID for a device and hides the device UUID.

If you are trying to do iBeacons, you're going to need to parse the Beacon UUID, Major ID and Minor ID from the manufacturer data in the [advertising data](https://github.com/don/cordova-plugin-ble-central#advertising-data). Note that the advertising data is different between iOS and Android. You can get beacon information with this plugin, you just need to write a lot of code.

Alternately you can look at an iBeacon specific plugin like https://github.com/petermetz/cordova-plugin-ibeacon

closed 2015-09-17 by don