don/cordova-plugin-ble-central #70: No api to read descriptor of characteristic

PeacePan, 2015-06-25:
I want to read the Characteristic Presentation Format Descriptor(0x2904) of Battery Level(0x2A19).

But there no any method to read descriptor.

don, 2015-06-29:
There's no API for descriptors at this time. It would be a nice enhancement in the future.

PeacePan, 2015-06-30:
I got it.
Thank for your response.
This is a good plugin.

closed 2015-06-30 by PeacePan

dnat112, 2015-08-23:
Hello,
I also would really like the ability to read and write to a descriptor. Is anyone currently working on this? I would be happy to work with someone to get this feature implemented.
Thanks.

don, 2015-09-19:
No one is working on this. Reading descriptors would be nice. What are your use cases for writing descriptors?

drewcovi, 2018-01-26:
I'm in need of this, and have used it extensively in the evothings plugin. It's extraordinarily helpful to add context to a characteristic value. consider it metadata for any given property. otherwise there's no graceful way to establish range, human readable label for the value, etc.

don, 2018-06-08:
See #111