don/cordova-plugin-ble-central #71: startNotifications seems to stack on Android when Arduino is turned off (then back on) and phone is reconnected

alexissantos, 2015-07-07:
On Android, when the Bluetooth connection is terminated by turning the Arduino off, connecting again seems to cause startNotifications to stack. If I disconnect properly by using `ble.disconnect`, there's no issue.

For example, when I first connect my phone to my Arduino, data sent via the rxCharacteristic appears as it should. Eg:

1. I transmit some text via the serial monitor send field on my computer
2. In my Meteor terminal, the data appears once.

[image]

However, if I turn the Arduino off and make sure that my phone has been disconnected, then reconnect after turning it on again, the following happens:

1. I transmit some text via the serial monitor send field on my computer
2. In my Meteor terminal, the data appears several times. (One time for each time it has been connected. If I connected seven times total, it could be shown seven times. For example, the gif below shows it on the second connect.)

[image]

## Phone Details
- One Plus One (64GB)
- Cyanogen OS 12.0-YNG1TAS2I3
- (Android 5.0.2)

## Arduino Details
- Arduino: Adafruit Metro (https://www.adafruit.com/products/2488)
- Bluefruit LE nRF8001: https://www.adafruit.com/products/1697

## Computer
- Mac OS X Yosemite 10.10.4

---
Let me know if you need any more info!

alexissantos, 2015-07-08:
Just updated the issue with a note on the version of Meteor I'm using, and the current workaround I'm relying on.

charlesalton, 2015-08-13:
I have been able to recreate this same issue under cordova with a TI BLE chip using a Moto X phone. Each time the peripheral device drops connection and is reconnected an extra notification listener is added causing a stack of responses at once. Killing the app clears the notifies but this seems to be the only solution right now to recover from a lost connection to a device and preventing duplicated notifications. I've tried the stopNotification and it doesn't work and disconnect doesn't work after the fact of a failed connection.

charlesalton, 2015-08-14:
The work around/fix I have found is modifying the "Peripheral.java" file in the plugins\com.megster.cordova.ble\src\android directory and adding the following two lines after "connectCallback = null;" around line 200 in the onConnectionStateChange function.

gatt.close();
gatt = null;

Then to get your project to recompile with this new code at least in cordova, I had to remove the directory "android" under "platforms" and then instruct cordova to add android as a new build platform.

don, 2015-09-17:
@charlesalton thanks for tracking this down, your suggested fix works great

closed 2015-09-17 by don

alexissantos, 2015-09-17:
@don Sweet! Thanks so much for taking care of this, Don! And you too, @charlesalton!

don, 2015-09-17:
@alexissantos I should have this pushed to npm by tomorrow

commit 38535a91 (2015-09-17): close gatt when device disconnects fixes #71