don/cordova-plugin-ble-central #72: Android Lollipop 5.0.1

vash15, 2015-07-16:
Hi guys! I'm using Android 5.0.1 but does not discover any devices. If I run my application with Android 4.4.3 it's ok.  Any idea?

don, 2015-07-17:
Are you filtering the scan data or scanning for all devices? In my experience filtering scan data on Android is bad, but generally works better with newer releases of Android. You're seeing the opposite, so maybe it's a 5.0.1 specific issue?

Andy-Donegan, 2015-07-17:
Hi Vash, 
I have tested on devices from 4.4.3 upwards and have it working on all devices.
But I have found certain devices which do not work or have issues with BLE on different Android Platforms.
Is this just happening with one device or many ? maybe look up your device and look for others having issues with BLE on the Android. 
Andy.

vash15, 2015-07-24:
I have this problem when scanning without filters. My device is a LG Magna 5.0.1. I tried to turn off the WiFi and clear the Bluetooth cache but it wont solve my problem. The ble device doesn't appear in the list.
This week I tried with a Samsung S6 Android 5.0.2 and it works fine! 
Is 5.0.1 definitely broken?

vash15, 2015-08-21:
I tested it on other devices and the Bluetooth works. It is likely that the problem is related to that specific device. Thanks a lot

closed 2015-08-21 by vash15