don/cordova-plugin-ble-central #73: Android random disconnects

thomasbaggio, 2015-07-29:
Hi Don 
we are using in our project your library.

I want to submit to your attention a list of Android BLE issues:
https://github.com/iDevicesInc/SweetBlue/wiki/Android-BLE-Issues 

We are facing many of these, in particular: 
"Random disconnects happen. They happen a lot. In some situations you can expect a disconnect every few minutes regardless of signal strength. You need to have very robust reconnect logic for user experience not to suffer."

We have also connection problems with some Lollipop devices. I don't know if these links can help:
http://stackoverflow.com/questions/28018722/android-could-not-connect-to-bluetooth-device-on-lollipop
http://stackoverflow.com/questions/21398766/android-ble-connection-time-interval?lq=1 

Thanks a lot,
Thomas

don, 2015-08-10:
@thomasbaggio since random disconnects are dependent on the phone, os, and device you're connecting to, I think the reconnect logic is best handled by the application, rather than the plugin. 

The failure callback of the [connect function](https://github.com/don/cordova-plugin-ble-central#connect) will be called whenever a peripheral disconnects. You can add reconnect logic in this event handler.

thomasbaggio, 2015-08-26:
I've read that turning off and on Bluetooth on Android (4.4.4 on Nexus and other) could improve the connection. Do you think it would be possible to insert these lines to add the enable and disable options? (without user confirmation)

public void enableBT(View view){
    BluetoothAdapter mBluetoothAdapter = BluetoothAdapter.getDefaultAdapter();
    if (!mBluetoothAdapter.isEnabled()){
        mBluetoothAdapter.enable();
    }
}

public void disableBT(View view){
    BluetoothAdapter mBluetoothAdapter = BluetoothAdapter.getDefaultAdapter();
    if (mBluetoothAdapter.isEnabled()){
        mBluetoothAdapter.disable();
    }
}

don, 2015-08-27:
@thomasbaggio Unfortunately the user needs to be prompted whenever Bluetooth is enabled or disabled.

Your code example might work but the Android docs warn "Bluetooth should never be enabled without direct user consent." So this isn't something I'd want to have in a generic plugin.

http://developer.android.com/reference/android/bluetooth/BluetoothAdapter.html#enable()
http://developer.android.com/reference/android/bluetooth/BluetoothAdapter.html#disable()

thomasbaggio, 2015-08-28:
@don Thanks! I agree with you.

closed 2015-09-11 by don