don/cordova-plugin-ble-central #74: Question Lightblue Bean

Arcrueid, 2015-07-31:
Hello!

I've been trying to use the write method with a bean lightblue but I can not send any information.

I can subscribe to the service and read without problem.

using
         service_uuid = 'A495FF10-C5B1-4B44-B512-1370F02D74DE'
         characteristic_uuid = 'A495FF11-C5B1-4B44-B512-1370F02D74DE'

There will be some detail I'm missing? since the bean Serial nothing comes of information.

don, 2015-08-10:
Are you using iOS or Android? Do you have some code? Check the properties on the characteristic, iOS is picky about write vs writeWithoutResponse.

don, 2015-09-17:
Those look like the correct ids. The tricky part is figuring out the bean protocols. I'd look at the node library to see how things work https://github.com/PunchThrough/ble-bean and translate that to something that works with cordova.

closed 2015-09-17 by don