don/cordova-plugin-ble-central #77: Is it possible to get the device MAC address on iOS?

objectkuan, 2015-08-04:
It seems that we can get the UUID on iOS and MAC address on Android.

Can we get the MAC address on iOS or UUID on Android?

Or any other parameter that can identify one device on different platforms?

Thanks

claudiobadii, 2015-08-06:
I have your same problem. On the console after the scan of beacons there is:

2015-08-06 16:32:37.756 NameOfApp [544:60b] {
    "Device Information" = <b9506ff2 27dbb6b9 506ff2>;
}
2015-08-06 16:32:37.758 NameOfApp [544:60b] Discovered {
    advertising =     {
        kCBAdvDataChannel = 37;
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataLocalName = estimote;
        kCBAdvDataServiceData =         {
            180A =             {
                CDVType = ArrayBuffer;
                data = "uVBv8ifbtrlQb/I=";
            };
        };
    };
    id = "BCBE4566-47A9-64D1-560C-C0D95B423CB9";
    name = estimote;b9506ff2 27dbb6b9 506ff2    rssi = "-95";
}

and if we look at the device information field, it contains the MAC, the TX, the Major and the Minor:

"Device Information" = <b9506ff2 27dbb6b9 506ff2>;

db:27:f2:6f:50:b9 is the MAC,
b6 is the tx (16*11 + 6 = 182, -256 + 182 = -74 dB),
50b9 is the major
f2f6 is the minor

So from the plugin how it is possible to retrieve this information?

Thanks

objectkuan, 2015-08-06:
Hi @claudiobadii , that's interesting. How do you get such device information? On iOS?

All information I get is that between `Discovered {` and `}`

claudiobadii, 2015-08-06:
Hi @objectkuan. Yes, on an Iphone 5s with IOS7.1 this information is sended by two iBeacon, but it is not retrieved by the plugin: only logged in the console.

don, 2015-08-10:
iOS hides the MAC address of the device and generates a UUID for you. Android allows access to the UUID. It's possible there is a way to get the MAC address from iOS based on what @claudiobadii is doing. If you find a generic solution that doesn't upset the Apple app approval process, I'll accept a pull request.

objectkuan, 2015-08-11:
Thanks @don and @claudiobadii .

If Android allows access to the UUID, I'll try to simply modify the plugin to do that for urgent production requirement. @claudiobadii provides a nice solution and we consider adding such feature to our next version of hardware.

I didn't know about the Apple issue, but I'll keep looking for an elegant solution to that.

holla2040, 2015-09-10:
Objectkuan,
did you find an elegant solution? I'm at a point where you where a month ago. My app is kind of broken because of this UUID/MAC issue. runs fine on android using MAC addresses. 
Craig

don, 2015-09-17:
@claudiobadii I'm not seeing "Device Information" when I scan devices. Is this another app reporting data from the Device Information service (0x180a)? Or data from within the Manufacturer Data field? I'm guessing this is some specific info that your device reports, rather than something generic we get for all iOS devices.

@objectkuan were you able to make a patch that does what you need?

@objectkuan @holla2040 why do you need the MAC address anyway? Can't you use the UUID generated by iOS for that device?

FWIW, On OS X there's a file ,/Library/Preferences/com.apple.Bluetooth.plist, that maps UUIDs to MAC addresses. AFAIK you won't see a mapping there until you've connected. It's possible there's something similar on iOS. You can look noble to see how they use the plist https://github.com/sandeepmistry/noble/blob/64053ad35a61b1101a0f9fd8c6f0abb5f03db684/lib/mac/uuid-to-address.js

holla2040, 2015-09-17:
For my use case, I have to match the physical product serial number (going to use the MAC address) with an online account. During commissioning, my use case might have 20+ unprovisioned units on at the same time, so the MAC address via advertisement is very simple to get the right unit matched with the right account. 

If I had the most common use case, single phone with single unit, then UUID is fine. My product is little more complicated. 

Yes, there are a work arounds, but ios is now special cased. My architecture isn't quite as simple now.  Its all good, my customer is sooo thrilled. His industry has NOTHING like this. I showed him an iPad demo last night, you should have seen his grin.

claudiobadii, 2015-09-18:
@don The "Device Information" are reported on console when I start the scan with your plugin. The device that I try to scan are Estimote Beacons (http://estimote.com/) and they give me this information.

objectkuan, 2015-09-18:
Hi all, 

Sorry for the delay. I think the problem we meet is "Getting UUID of the BLE Device on Android".

I've tried different techniques on Android, and currently the furthest solution I get is [here](http://stackoverflow.com/questions/22016224/ble-obtain-uuid-encoded-in-advertising-packet#).

But in my case, I cannot find a record with type 0x07, which stands for UUID according to the answer.

I am still searching for a graceful method. Looking forward to any good advice.

don, 2015-09-18:
@objectkuan you're not going to be able to get a UUID on Android. The UUID on iOS is generated by the iOS device. Different iOS devices will get different UUIDs for the *same peripheral*.

The MAC address is usually based on the hardware, but it doesn't have to be. Gimbal beacons for instance change their mac address with every advertising packet.

don, 2015-09-18:
@claudiobadii I see what that Device Information message is coming from now.

Device Information (0x180A) is in advertising information that the Estimote sends out. You access this JavaScript.

	if (peripheral.advertising 
		&& peripheral.advertising.kCBAdvDataServiceData 
		&& peripheral.advertising.kCBAdvDataServiceData["180A"]) {
	
		var serviceData = new Uint8Array(peripheral.advertising.kCBAdvDataServiceData["180A"]);
	
		// print out the bytes
		for (var i = 0; i < serviceData.length; i++) { 
			console.log(serviceData[i].toString(16)) 
		}

	}

don, 2015-09-18:
Summary - iOS doesn't let you get the MAC address of a device, it gives you a random UUID instead.

Some devices, like Estimote, may give you the MAC address (or other identifier) in the Advertising information or through a Bluetooth service.

closed 2015-09-18 by don

shubham-santuwala, 2016-04-08:
Hello sir,
I'm new to Development, and Even I'm also facing the same UUID/MAC address problem , don't know how to come across with this problem and my boss wants it.
So if you guys having any sorts of information to come across from this so plz let me know ASAP.

don, 2016-04-08:
@shubham-santuwala you can't get the MAC address on iOS, your phone generates a UUID and hides the MAC address.

chitranshmedsolis, 2016-12-13:
@don : Sir..! Can we store this UUID for our app purpose. Means would this approach not considered as Apple Violence?

Thanks & Regards,
Chitransh Mathur

don, 2017-01-05:
@chitranshmedsolis you can store the UUID on the phone, but it will be unique to that phone.

If we both have iPhones and scan the same peripheral, we'll see different UUIDs. iOS generates the UUID on the device and hides the MAC address.

chitranshmedsolis, 2017-01-05:
`You can filter the scan results by service UUID. This doesn't work well on Android so I usually manually filter the results before presenting to the user.`

Actually in iOS when any device is connected & it's characteristic found then we read the measurement of that device so it is returning us the results.

`There's usually some way to work around it but it's always implementation specific.`

You are correct. I need to debug those thing by which I'm getting last readings from devices.

But you know it is happening only with AnD BLE devices not with CONTEC devices.

Thanks :)

faytom, 2018-01-26:
@don I am working on a project which requires to connect to two different ble devices in really short period of time while these devices are awake. So the app will periodically scan for these two different devices and try to connect to them.

On iOS, once the device has been found separately, I will store their UUID, then later I can connect to them with their UUID without additional scanning. It works pretty good without any problem.

However, this doesn't work on Android. I start scanning for the first ble device and store its MAC address, then scanning for the second ble device and also store its MAC address. Now, if I try to connect to the first device with its MAC address attached, it just won't work. It seems that, if you start scanning again, the previous discovered devices will be cleaned. 

Is there any way to get around it? I also try to scan for two devices together with their own service UUIDs, but nothing returned.

beesop, 2020-02-27:
If the MAC is hidden by iOS and the alternative is to use the UUID, which in iOS is different on each connection... then, what method is recommended to identify a device uniquely in IOS? the device name maybe?

My app needs to identify a BLE remote control, if the hardware is the original brand created for the app, then it would allow its use, otherwise it should not allow it. My original idea was to use the MAC, so that I can modify the firmware on the remote control so that all units have the same series that identifies the products of my brand. If this is not possible in iOS, which alternative would be the most recommended?

Thank you

vivekgajbe, 2020-09-07:
iOS can make connection with UUID
and android can make make connection with MAC address 

is there any way to connect my BLE device ,in both platform ,with same unique id ?(for same login/user)

yassinemenssi, 2021-12-13:
You can access to the MAC ADDRESS without problem in iOS 12. To get the mac address you have to follow the next steps.

Parse the Data received by the BLE device to String.
extension Data{
func hexEncodedString() -> String {
        let hexDigits = Array("0123456789abcdef".utf16)
        var hexChars = [UTF16.CodeUnit]()
        hexChars.reserveCapacity(count * 2)

        for byte in self {
            let (index1, index2) = Int(byte).quotientAndRemainder(dividingBy: 16)
            hexChars.insert(hexDigits[index2], at: 0)
            hexChars.insert(hexDigits[index1], at: 0)
        }
        return String(utf16CodeUnits: hexChars, count: hexChars.count)
    }
}

Add a separator ":" to the address.
extension String {
    func separate(every stride: Int = 4, with separator: Character = " ") -> String {
        return String(enumerated().map { $0 > 0 && $0 % stride == 0 ? [separator, $1] : [$1]}.joined())
    }
}
In didReadValueForCharacteristic( characteristic: CBCharacteritic) you can use the previous 2 functions to get the mac address.
func didReadValueForCharacteristic(_ characteristic: CBCharacteristic) {
if characteristic.uuid == BleDeviceProfile.MAC_ADDRESS, let mac_address = characteristic.value?.hexEncodedString().uppercased(){
            let macAddress = mac_address.separate(every: 2, with: ":")
            print("MAC_ADDRESS: \(macAddress)")
        }
}

Crl555, 2022-11-23:
@vivekgajbe did you ever solve for this? I have the same problem I am trying to solve for