don/cordova-plugin-ble-central #78: Anyone had any issues with iOS 9?

markygab, 2015-08-06:
I'm just about to check this out as app users with iOS 9 Beta are reporting that the app crashes immediately that they attempt to use BLE.

OnkarBhattacharya, 2015-08-13:
Issue 1 : Property 'isConnected' not found on object of type 'CBPeripheral'
Issue 2 : Property 'UUID' not found on object of type 'CBPeripheral'.

Seems like both properties are removed. (Source : https://developer.apple.com/library/prerelease/ios/documentation/CoreBluetooth/Reference/CBPeripheral_Class/index.html#//apple_ref/occ/cl/CBPeripheral)

don, 2015-08-14:
This plugin will work with iOS 9 in the future. I haven't spent any time on iOS 9 compatibility yet.

thomasbaggio, 2015-08-17:
Do you plan to test iOS 9 compatibility before the official launch? We have an app that uses your plugin and we must publish it in September .Thanks a lot.

don, 2015-08-18:
@thomasbaggio it's in a long queue of stuff to be done, ideally before iOS 9 launches

thomasbaggio, 2015-08-20:
thanks! If I can .... do you plan to do also the Windows Phone version in the next months? (only to know)

markygab, 2015-08-24:
@don Would you be open to accepting payment to expedite this iOS9 work?

don, 2015-08-24:
@markygab sure reach out first + last at gmail

don, 2015-08-24:
@thomasbaggio Ideally I'll add Windows 10 and Windows Phone support, but it'll be a few months at least. The Windows scanning APIs are really crippled so it probably won't work the same.

thomasbaggio, 2015-08-31:
@don ok thanks! Take a look here.. you can use javascript instead of c#, it works for WP and also Windows:
https://github.com/randdusing/BluetoothLE/pull/151

don, 2015-09-02:
iOS9 support is merged in from #62 

The plugin has been migrated to NPM. Use the new id `cordova-plugin-ble-central`

closed 2015-09-02 by don

markygab, 2015-09-03:
Thanks Don,

We got it all working today. I might send you you through some our implementation to take a look at tomorrow if that's still ok?

Cheers,
Mark

morganRotaStabelli, 2015-10-01:
damn! I have gone crazy today to implement the same changes , fool that I did not before ensuring that you have already been resolved ... however I found a difference , I have implemented
( peripheral.state ! = CBPeripheralStateDisconnected )
instead
peripheral.isConnected
Do you think is a mistake ?

wlzimmer, 2015-11-18:
Do we know when the cordova ble plugin will support ios 9.x???   If not, is there another library that has been tested with ios 9.x??

wlzimmer, 2015-11-18:
Sorry I missed the Sep 2 notes.

don, 2015-11-18:
This plugin works with iOS 9. If you are seeing any problems, please open a new issue with details and a stack trace.