don/cordova-plugin-ble-central #79: ble is not defined

daniperezluna, 2015-08-07:
Hi, I'm triying to develop an app with ionic an ngCordova but when I launch the app I receive the follow message: 

ReferenceError: ble is not defined
    at Object.isEnabled (ng-cordova.js:610)
    at new <anonymous> (controllers.js:35)
    at invoke (ionic.bundle.js:13012)
    at Object.instantiate (ionic.bundle.js:13020)
    at ionic.bundle.js:17289
    at self.appendViewElement (ionic.bundle.js:48435)
    at Object.switcher.render (ionic.bundle.js:46629)
    at Object.switcher.init (ionic.bundle.js:46549)
    at self.render (ionic.bundle.js:48295)
    at self.register (ionic.bundle.js:48253)

NewLunarFire, 2015-08-07:
ble is not defined if the platform you are testing your code on doesn't support BLE.

daniperezluna, 2015-08-07:
I am testing my code directly on Android 4.4.2 who supports BLE.

daniperezluna, 2015-08-10:
If you need more information about my devices I am using :

NewLunarFire, 2015-08-10:
From the specs of the phone Bluetooth Smart should be supported, but just to make sure have you been able to connect to your peripheral from another application?

don, 2015-08-10:
`ble` is defined by the plugin. If you're seeing errors about ble not defined, it's typically that the plugin didn't install correctly or you're trying to call a function before Cordova fires `ondeviceready`. Try uninstalling and re-installing the plugin first.

daniperezluna, 2015-08-11:
I install again the plugin and it's works properly. I am using ionic and I can scan but I can't connect to my device. I have the macAddress but when I do ble.connect(peripheral.id, app.onConnect, app.onDisconnect); never connect to the device.

Someone knows why is it?

I have tried with the heart rate example, using the functions inside ionicPlatform and I have changed the values of the heartRate.service and heartRate.measurement.

I have a blood pressure monitor and those values are '1810' and  '2a35' but when the monitor sends data and I try to connect always say that is not possible.

I saw you use ble.notify but it is not present in factory.

don, 2015-08-14:
@daniperezluna try connecting to your phone to the monitor using [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en)

You'll need ble.startNotification to get blood pressure measurement, since that's sent with an indication and not readable.

soumendub3, 2015-09-16:
@don I have the similar issue as *daniperezluna* started the issue with and I followed the instructions, uninstalled the plugin and reinstalled back but I cannot get the plugin to work. I am using ionic framework with AngularJS. 
@daniperezluna Also, what I am not following is how do you get $cordovaBLE in controllers.js? Are you still using the same code as posted above? If not, can you share your working code? Thanks.

daniperezluna, 2015-09-16:
@soumendub3 Have you tried to debug it on your phone? My problem was that this message appears when I open the app in the browser

don, 2015-09-17:
@soumendub3 I suggest you get simple Bluetooth examples working without Ionic or Angular, then work on adding Bluetooth support to your Ionic app.

I have a simple Ionic Bluetooth LE App you can try too. https://github.com/don/ionic-ble. Note that it does *not* use the ng-cordova wrappers.

Look at BLE in [services.js](https://github.com/don/ionic-ble/blob/455e4cb618b7f8b234b3ab4832c66f4f353ba2f4/www/js/services.js#L52) and BLECtrl and BLEDetailCtrl in [controllers.js](https://github.com/don/ionic-ble/blob/455e4cb618b7f8b234b3ab4832c66f4f353ba2f4/www/js/controllers.js#L16)

soumendub3, 2015-09-17:
@don Thanks for the response Don - yes, I found your ionic-ble code and I started with your example. I had trouble with it as well and I created an issue in the project. I understand the BLE and BLEDetailCtrl and see how you defined them but my trouble there was different. Can you please take a look at that one? [Link to the issue](https://github.com/don/ionic-ble/issues/2)

soumendub3, 2015-09-17:
@daniperezluna Yes, I tried on my phone and it was not related to browser.

don, 2015-09-17:
@daniperezluna great news

@soumendub3 I'll continue this in the other project

closed 2015-09-17 by don

daniperezluna, 2015-09-17:
@soumendub3 Here you are my code https://github.com/daniperezluna/BlueHeart It's an app to read data from a Blood Pressure.

wifixcort, 2016-02-21:
I have the same issue in ionic. If I write:
`
ble.isEnabled(function() {
     alert("Bluetooth is enabled");
      ble.scan([], 5, function(device) {
         console.log(JSON.stringify(device));
}, function(function(error){}){});     
        },
      function() {
           alert("Bluetooth is DISABLED");
       }
    );
`
works, but just in app.js, if I try to use in my controller.js it doesn't work

abnuozal, 2016-03-23:
Make it like this in your controller:

`   $ionicPlatform.ready(function(){
     window.ble.isEnabled(
      //do something
);
   })`

20gg, 2017-01-03:
@wifixcort  you can try this way.bind Event with $scope ,just like 

> $scope.onIsEnabled = function(){
    ble.isEnabled(
    function() {
        console.log("Bluetooth is enabled");
    },
    function() {
        console.log("Bluetooth is *not* enabled");
    }
     );
  }

wifixcort, 2017-01-03:
@20gg Thanks for your advice, I'll try it :+1: