don/cordova-plugin-ble-central #80: Communication with Microchip RN4020

JasaNiklanovic, 2015-08-19:
Hey,

does anybody have any experience/ideas regarding communication with Microchip RN4020 bluetooth low energy chip? im interested in Android/iOs
http://www.microchip.com/wwwproducts/Devices.aspx?product=RN4020

Thanks, Jasha

don, 2015-08-19:
As long as you have a Bluetooth Low Energy service running on the RN4020, you should be able to use this plugin to communicate with it using Cordova on Android and iOS.

Janckk, 2015-08-19:
Actually I'm running into the same problem as those microchips are using some kind of "MLDP" protocol or data structure.

What I was able to do was to connect to the bluetooth. But I have no idea what kind of data to send as it doesn't seems to be picking it up. I'm trying to send everything from typed arrays to binary data. 

Does anyone have experience with this?

Janckk, 2015-08-21:
Ok, I'm debugging what I'm sending and what is being send by the microchips example application that works.

This is what I'm currently sending:
[image]

And this is what I'm trying to achieve:
[image]

The main difference seems to be the "R" that is being send with the request. How can I get rid of it?

don, 2015-08-24:
@Janckk [write](https://github.com/don/cordova-plugin-ble-central#write) sends a write request. [writeWithoutResponse](https://github.com/don/cordova-plugin-ble-central#writewithoutresponse) sends a writeCommand.

You might need to manually add the R in your code. Typically I need a combination of vendor documentation and debugging to get some devices to work.

closed 2015-09-17 by don