don/cordova-plugin-ble-central #82: Characteristic writes and notifies?!

RomainGoncalves, 2015-08-24:
Hi,

I'm trying to subscribe and write to the same characteristic and service. Is that normal?
In the peripheral data I get:

````
{
      "service": "53455206-444C-5020-4E49-52204E616E6F",
      "isNotifying": false,
      "characteristic": "4348411D-444C-5020-4E49-52204E616E6F",
      "properties": [
        "Write"
      ]
    },
    {
      "service": "53455206-444C-5020-4E49-52204E616E6F",
      "isNotifying": false,
      "characteristic": "4348411D-444C-5020-4E49-52204E616E6F",
      "properties": [
        "Notify"
      ]
    }
````

And when I use startNotification on that characteristic, I receive an "Unknown error" in the failure callback.

Is it normal to have a characteristic that writes and notifies?

Thanks for any help.

don, 2015-08-24:
It's fine to have notify and write on one characteristic. I'm not sure why it's being listed 2x. I'd expect the characteristic to be listed like 

    {
      "service": "53455206-444C-5020-4E49-52204E616E6F",
      "isNotifying": false,
      "characteristic": "4348411D-444C-5020-4E49-52204E616E6F",
      "properties": [
        "Write", "Notify"
      ]
    }

Did you create the service on the iPhone? Or is there an app I can download to duplicate this?

RomainGoncalves, 2015-08-25:
Hey,

thanks for looking into this.

I didn't create anything. I have a physical device that has Bluetooth capabilities and I'd just like to access it. On the iPhone i'm just using your plugin and a bit of javascript with AngularJS.

But once again, with the other apps, it works just fine...

don, 2015-08-25:
The plugin caches the characteristics when they are discovered. It looks like iOS is reporting 2 characteristics with the same UUID but different properties for your device. I've never seen this, so I'm not sure if it's a device issue or iOS issue.

What version of iOS are you running?
What's the bluetooth device you have? I'd like to see if I can duplicate this. Is this something I can buy?

If the plugin gets an error you could look up the characteristic again, but it would be better to see if this could be handled during the initial discovery.

Does LightBlue list the characteristic 1x with both read and write or does it list it 2x with one property each?

RomainGoncalves, 2015-08-26:
Hey,

so LightBlue discovers 2 characteristics with the same UUID, but one WRITE and the other NOTIFY. On LightBlue, I can subscribe to the NOTIFY char, then WRITE to the 'other' char, and get the notification back.

I'm running the latest iOS for now so 8.4.1.
The device is the [Texas Instrument DLP Nano](http://www.ti.com/tool/dlpnirnanoevm), I'm not sure you'd want to buy it, it's a bit pricey.

I've looked a bit more at the Objective-C code (which I don't know at all), and I think in the 'method' (?) ```decodeCharacteristicProperties``` it reads all characteristics, one at a time, and creates a new entry for each. So, apparently, as the device broadcasts 2 chars with the same UUID, it creates 2 entries. But once I try to subscribe, it find the first entry (by its UUID), but finds that it's a WRITE char. And it fails.

Maybe in the method ```startNotification``` it would be helpful to add a condition that checks if the char has the NOTIFY property also. If not, it will not return, and will continue to find the second one. 
I tried doing that in Xcode, but somehow it doesn't work...

Thanks again for your help

RomainGoncalves, 2015-08-27:
OK, so:

I've worked it out. Made something that's not really pretty though. I had to make a separated getCommand and findCharacteristicFromUUID methods, apply them to startNotification. And in findCharacteristicFromUUID I added an IF statement filtering the property associated to the char, making sure it was NOTIFY.

It works but once again it's ugly. I received work from the manufacturer and they said that they did it on purpose as the WRITE command uses small packets but the NOTIFY uses a large amount of packets and they wanted to use different stuff on either. So no choice there. Maybe good to add a small implementation of that case in the plugin.

Here's the changed findCharacteristicFromUUID method:
```objective-c
// Specific for notification
-(CBCharacteristic *) findCharacteristicFromUUIDNotification:(CBUUID *)UUID service:(CBService*)service
{
    NSLog(@"Looking for %@", UUID);
    for(int i=0; i < service.characteristics.count; i++)
    {
        CBCharacteristic *c = [service.characteristics objectAtIndex:i];
        NSLog(@"Characteristic %@", c);
        if ([self compareCBUUID:c.UUID UUID2:UUID]){
            NSLog(@"Properties %lu", c.properties);
            if(c.properties == 16){
                return c;
            }
        }
    }
    
    return nil; //Characteristic not found on this service
}
```

I can fork, PR if you're interested... but again, there might be a cleaner way to go about it. I don't know Objective-C at all.

Cheers and thanks for that plugin

don, 2015-08-27:
I'm surprised the manufacturer said they did this on purpose. They should have used two characteristics. I'd think that two characteristics with the same UUID breaks the spec, but I'll need to check on that.

The patch fixes your case, but `write` only works for you since they define write first. If they change the order of the characteristics on the peripheral, write will break. 

You should also check if the properties contains Notify, rather than == 16.

    if (c.properties & CBCharacteristicPropertyNotify) != 0x0) {
        return c;
    } 

If you're going to make a patch, it should handle indicate too.

Although I don't like it, I suppose the property check is not expensive. Maybe findCharacteristicFromUUID could take an optional property and filter the characteristics like your example.

don, 2015-08-27:
@RomainGoncalves it looks like it's legitimate. "A service may have multiple characteristic definitions with the same Characteristic UUID." (Bluetooth Specification Version 4.2 [Vol 3, Part G] section 3.3.1 Characteristic Declaration)

I need to add a filter to *all* characteristic lookups in a future version.

don, 2015-09-17:
@RomainGoncalves the latest code from git should work for your device now. It will be in v1.0.1 once that's released. Until then then you can install from git.

    cordova plugin add https://github.com/don/cordova-plugin-ble-central.git

closed 2015-09-17 by don