don/cordova-plugin-ble-central #83: 128-bit UUID support

airbly, 2015-08-25:
Hi,

It does not appear that this library supports 128-bit UUIDs.  Is that correct?  If I connect to a device that has several custom services and characteristics with 128-bit UUIDs defined, they call come up with a value of "0000"

Here is an example of a response from a connection.

{"characteristics":[{"characteristic
":"2a00","service":"1800","properties":["Read"]},{"characteristic":"2a01","servi
ce":"1800","properties":["Read"]},{"characteristic":"2a04","service":"1800","pro
perties":["Read"]},{"descriptors":[{"uuid":"2902"}],"characteristic":"2a05","ser
vice":"1801","properties":["Read","Indicate"]},{"characteristic":"2a06","service
":"1802","properties":["WriteWithoutResponse"]},{"descriptors":[{"uuid":"2904"},
{"uuid":"2902"}],"characteristic":"2a19","service":"180f","properties":["Read"]}
,{"characteristic":"0000","service":"0000","properties":["Read"]}]

Note this section:
{"characteristic":"0000","service":"0000","properties":["Read"]}

It should read:

{"characteristic":"12340000-0010-0080-0000-805F9B34FB00","service":"12340000-0010-0080-0000-805F9B34FB00","properties":["Read"]}

Thank you.

- Chris

don, 2015-08-25:
This library supports 128 bit UUIDs.

Are you seeing this on iOS or Android? What version?

See if your custom service appears correctly using other tools like LightBlue or nRF Master Control Panel. Ensure that the GATT info is correct for your custom service. I've see problems with peripherials sending bad data when there errors in code or configuration.

airbly, 2015-08-26:
Hi Don,

Thanks for getting back to me.  Your project is quite helpful.  I did some more testing with Android 4.4.2 vs iOS 8.4.1.  Below are the results of those tests.

================================================================
On an iOS device (iPad mini 2), things look great.  A ble.connect function call will return my custom UUID as seen below.  This shows the complete UUID.

{
  "service":"12345678-0008-0000-0805-F9B34FB00000",
  "isNotifying":false,
  "characteristic":"87654321-0000-0805-F9B3-4FB000000000",
  "properties":["Read"]
}

and I am able to read the value of that characteristic with:
  ble.read(bleConnectedId,"12345678-0008-0000-0805-F9B34FB00000", "87654321-0000-0805-F9B3-4FB000000000",bleReadSuccess, onError);

================================================================
Android however works a bit differently.  On my Galaxy Tab 4 running Android 4.4.2, this is the response I receive for the same service from a ble.connect() call:

{
  "characteristic":"0000",
  "service":"0000",
  "properties":["Read"]
}

Interestingly however, I am able to read this characteristic using ble.read and the 0000 and 0000 values.

ble.read(bleConnectedId,"0000", "0000",bleReadSuccess, onError);

The problem I have with this is I am unable to really determine that the characteristic is the one I want.  The 16-bit UUID value of 0000 does not mean much when it's real 128-bit UUID is 87654321-0000-0805-F9B3-4FB000000000.
================================================================

Any thoughts or solutions you might have would be helpful.

Thank you.

- Chris

airbly, 2015-08-26:
Additionally, with Android 4.4.2, if I have two unique characteristics under one service, such as:

Service: 12345678-0008-1234-5F9B-34FB00000000
   Characteristic: 87654321-4321-5F9B-34FB-123412341234   UINT32
   Characteristic: 87654321-4321-5F9B-34FB-123412341235   UINT32

iOS displays them just fine:

{
  "service":"12345678-0008-1234-5F9B-34FB00000000",
  "isNotifying":false,
  "characteristic":"87654321-4321-5F9B-34FB-123412341234",
  "properties":["Read"]
},
{
  "service":"12345678-0008-1234-5F9B-34FB00000000",
  "isNotifying":false,
  "characteristic":"87654321-4321-5F9B-34FB-123412341235",
  "properties":["Read"]
}

However on Andriod 4.4.2, one of the characteristics is missing.  This is all that is shown.

{
  "characteristic":"0000",
  "service":"0000",
  "properties":["Read"]}
}

So you know, for brevity I have cut out the standard Generic Access and Generic Attribute services that GATT requires.  

I hope that helps.  I should also point out that I am using the version of your library that is included with the PhoneGap Desktop developer app from the App store / Google Play.  I am not sure how current the version included with that build is.  Is there a way to check the version number either by printing a variable or calling a function?

Thanks.

- Chris

airbly, 2015-08-26:
One last bit of information.  I am getting the data I am posting by calling console.log(JSON.stringify(data)).  If there is a better way to display the contents of the data created by the ble.connect() function, let me know and I'll try that as well.

Thanks again.  Sorry for all the messages.

- Chris

don, 2015-08-27:
Chris, the version in the PhoneGap Developer app is the latest. (I'm not sure how to check the version, but I haven't made any changes since adding it to PGDev app.)

I'm thinking this is an issue with Android 4.4.2 or a device specific problem. The plugin tries to take the data from Android and convert it to JavaScript. Can you try this on another device?

Does [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) show the correct 128 bit UUID?

airbly, 2015-08-27:
Hi Don,

Using the nRF Master Control Panel app on Andriod 4.4.2, I am seeing the correct and full 128-bit UUIDs. 

I will see if I can get a hold of a different version Android device or upgrade/downgrade one of the ones I have.  The two Android devices I have are running 4.4.2.

And I assume that you don't have a version variable or function exposed that would return the version of the library?  Something like ble.getLibVersion?

Thanks for your help.

- Chris

don, 2015-09-17:
@airbly were you able to resolve this with newer versions of Android or different devices?

to answer your other questions
 * stringifying the JSON returned from the connect success function is a good way to print out details.
 * ble.getLibVersion is a good idea, but it doesn't exist. For now, look at the PhoneGapDev version and compare it to their [github repo](https://github.com/phonegap/phonegap-app-developer)

Re-open this if you're still having issues.

closed 2015-09-17 by don