don/cordova-plugin-ble-central #88: $cordovaBLE.startNotification must be restarted

whittenjaw85, 2015-09-04:
Working similar to your BluefruitLE application, I am using the ionic platform with an angular twist. The issue I'm observing is that this function stops calling the callback after the first go.

  $cordovaBLE.startNotification(
        deviceId,
        bluefruit.serviceUUID,
        bluefruit.rxCharacteristic).then(
          onDataF,
          onErrorF
      );

If in the onDataF() call, I place this:

var onDataF = function(data){
    $scope.uartRx = bytesToString(data);
    console.log($scope.uartRx);
    onConnect();

  };

It will behave.
I thought it was strange enough to notify you. Thanks for the great work! I have thoroughly enjoyed tinkering with your library.

jonalling, 2015-09-05:
I am having the same issue with "startNotification".  I can only get one function call, even though my characteristic value is definitely changing.  I've implemented it a little differently than you though. I'm using in a service and with ngCordova's angular wrapper.

  this.notifyBLE = function(deviceId, data) {
    serviceUUID = data.service;
    characteristicUUID = data.characteristic;
    $ionicPlatform.ready(function(){
      var notify = $cordovaBLE.startNotification(deviceId, serviceUUID, characteristicUUID);
      notify.then(
        function(result) {
          value = new Uint8Array(result);
        },
        function(error) {
      });
    }); // end $ionicPlatform.ready
    return value;
  } // end notify function

don, 2015-09-11:
@whittenjaw85 calling onConnect seems like the wrong solution. Ideally you want to stay connected to the device passively receiving notifications.

It looks like the ngCordova wrapper not defined correctly. 

A promise should only ever be called *1x*. The callback for the notification need to be called *every time* a notification is received.

I suggest you get a simple notification project working without Angular, then work on a solution.

I created an example that uses the [bleno](https://github.com/sandeepmistry/bleno)' test service. You can run this test peripheral on OS X or Linux. Clone the repo and `node test.js`

create a new cordova project

    $ cordova create notify com.example.notify Notify

replace notify/www/js/index.js with https://gist.github.com/don/d9fef2fb45fbd67a372d

    $ cd notify
    $ cordova plugin add cordova-plugin-ble-central
    $ cordova platform add android
    $ cordova run --device

/cc @jonalling

don, 2015-09-16:
This is a problem with the ng-cordova wrapper. https://github.com/driftyco/ng-cordova/issues/981

closed 2015-09-16 by don

jonalling, 2015-09-16:
> This is a problem with the ng-cordova wrapper. driftyco/ng-cordova#981

Thanks Don.  I've been in contact with the Ionic guys since the ngCordova
wrapper and most of Angular is over my head.  I haven't been able to get a
response about this issue recently, but I forwarded your comments.

Jon

whittenjaw85, 2015-09-17:
@don. This does seem to be a problem. The ngCordova team is trying to wrap
your promise in another promise instead of simply wrapping the result in a
more angular style. I'll try to tinker with ngcordova to adjust how the BLE
central is wrapped. Thanks for the pointer.

don, 2015-09-17:
@whittenjaw85 thanks

jonalling, 2015-09-30:
@whittenjaw85 Any progress on the ngcordova wrapper for BLE? I'm really eager to get BLE working in Ionic.