don/cordova-plugin-ble-central #89: Dumb question: Can two phones exchange data?

daviddahl, 2015-09-04:
I would like to exchange a very short string between two phones over bluetooth, perhaps 100 chars.

don, 2015-09-11:
tl;dr this won't "just work" for you

This plugin will let you connect to a Bluetooth service on another phone. It allows your phone to act as a Central device. If there's a service on another phone that can send and receive strings you could make thiswork. You'd need the other phone to act as a Peripheral on offer a service. 

This plugin won't help you creating peripherals or services. iOS has good support for being a peripheral. Only the latest Android OS and hardware can be a peripheral. AFAIK not there's no cordova-plugin-ble-peripheral yet. (I'm working on one but it'll be a while.)

closed 2015-09-11 by don

daviddahl, 2015-09-11:
Thanks! I'll keep digging.