don/cordova-plugin-ble-central #91: error console android 5.1.1

KinG-InFeT, 2015-09-10:
hi i have this problem in console:
I/chromium﹕ [INFO:CONSOLE(173)] "No Content-Security-Policy meta tag found. Please add one when using the cordova-plugin-whitelist plugin.", source: file:///android_asset/www/plugins/cordova-plugin-console/www/console-via-logger.js (173)
09-10 17:55:18.609  14568-14634/test.bt D/PluginManager﹕ exec() call to unknown plugin: Console

and not find the bluetooth devices on my nexus 5 android 5.1.1.... fix?

don, 2015-09-11:
It looks like you're missing the cordova-plugin-console plugin or it's not installed correctly.

KinG-InFeT, 2015-09-11:
i installer this plugin with this command:

cordova plugin add cordova-plugin-console

and build with all plugins installed.... but not work....

don, 2015-09-11:
Something is making an `exec` call to Console which is missing

    09-10 17:55:18.609 14568-14634/test.bt D/PluginManager﹕ exec() call to unknown plugin: Console

Maybe it's your app? Maybe it's another plugin? The console plugin is nice to have but not required. You could try removing it.

Look for a `cordova.exec` call with `Console` as the 3rd argument in your code.

closed 2015-09-17 by don