don/cordova-plugin-ble-central #95: Possible to know if startNotification command succeeded, before getting notifications back?

poshaughnessy, 2015-09-17:
Hi.

I'd like to start notifications, then once notifications have been successfully registered, do a `write`. (The write command will prompt the peripheral to send a notification response).

Currently I'm not waiting on anything. I'm just calling `startNotification` and then calling `write` straight afterwards. That works most of the time. But occasionally `startNotification` seems to fail. In that case, we get the `failure` callback, so I can then retry `startNotification`. But in that case I don't want to have already sent the `write` command, because it will change the state, and I won't know what the response was... So ideally I'd only issue the `write` when I know `startNotification` was successful.

From the README, it sounds like that's what the `success` callback means:
 
    success: Success callback function that is invoked when the connection is successful. [optional]

But from testing with our app, it's actually invoked when we get a notification back?

Would it be possible to have a `success` callback to know if `startNotification` worked in the first place, separate to that?

I could have a look at doing this. But pretty new to Bluetooth, so wanted to check this makes any sense!

Thank you.

don, 2015-09-17:
That documentation in the README is incorrect. (I'll fix it.) The success callback is invoked only when there's a notification. Right now the only way to know the startNotification was successful, is that you don't get an error callback. In my experience, the error callback happens fast, like trying to start a notification on a characteristic that doesn't support notifications. Are you seeing this fail occasionally on iOS or Android? Do you know why it fails?

I think the right way to do what you want would be to have multiple success callbacks. One for startNotification success and another for the actual notification. (I do this in some other plugins, but it's sort of a clunky solution.)

poshaughnessy, 2015-09-18:
Great, thanks for updating the docs.

At the moment I'm thinking we probably need to add this other success callback, if possible. We could try this in our fork and see how it looks. I'm hoping it won't be too clunky; perhaps the existing 'success' callback could be renamed something like 'onNotification'?

I hope to gain more insight on Monday about why it might be failing, though. We'll hold off til then!

Thanks,

Peter

don, 2015-09-18:
@poshaughnessy see if you can get something that works for you. 

This change would break the existing API, so I'm not going to be too excited about adding this. I'd be interested, when this is breaking for you. Is there something in the code that can be improved? Or are some devices just slow to add notifications? Does this happen on iOS and Android? 

Maybe you can add another API call rather than modifying startNotification?
    ble.startNotificationWithConfirmation(notificationCallback, notifcationAddedSuccess, failure)

poshaughnessy, 2015-10-01:
@don Sorry for the delay in responding. 

To answer your question, I think I've only actually seen the setNotification failure callback logged on Android. I've seen it at least 3 times, but over the course of quite a lot of testing. So hopefully it's quite rare. (Then again, we've done more testing on iOS than Android).

We found a number of other bugs that were affecting our communication and with the other fixes it's working better now. I think we probably don't need to worry about this edge case just now, but I'd like to revisit it in our next development phase when we'll be trying to firm everything up for production. I'm still concerned that it could fail to set notifications and we want to handle it in a better way.

That's likely to not be for a few weeks now though. Would you be happy to leave the issue open until then? (If you'd rather close it for now, maybe I could ask to re-open it later?)

don, 2015-10-23:
@poshaughnessy I can re-open this issue later if you find it's necessary

woodz-, 2018-07-30:
Just a quick thought,

I am also looking for a callback of startNotification giving me a status whether it was set up successfully. Without having looked deeply into the code behind the scenes, a short view on the Java's BluetoothGatt lib tells that [setCharacteristicNotification](https://developer.android.com/reference/android/bluetooth/BluetoothGatt#setCharacteristicNotification(android.bluetooth.BluetoothGattCharacteristic,%20boolean)) returns a boolean true, 

> if the requested notification status was set successfully

Would this be enough to proof successful notification setup? Wouldn't it be possible to yield this Java return value into a js callback?