don/cordova-plugin-ble-central #96: iOS9 and xCode7 compiling issue

jonalling, 2015-09-17:
Cordova and Ionic have all been updated to latest releases, but I am getting this issue when trying to build the project in xCode. It specifically only seems to fail on the cordovaBLE plugin. If I remove the plugin, it builds with no errors.

** BUILD FAILED **

The following build commands failed:
CompileC build/cordovaDev.build/Debug-iphonesimulator/cordovaDev.build/Objects-normal/i386/BLECentralPlugin.o cordovaDev/Plugins/com.megster.cordova.ble/BLECentralPlugin/BLECentralPlugin.m normal i386 objective-c com.apple.compilers.llvm.clang.1_0.compiler
(1 failure)
Error code 65 for command: xcodebuild with args:

don, 2015-09-17:
Try version 1.0.0 of the plugin from npm. https://www.npmjs.com/package/cordova-plugin-ble-central

Note that you should use the new plugin id cordova-plugin-ble-central instead of com.megster.cordova.ble.

jonalling, 2015-09-17:
Yes, I added the plugin back in after removing it using the new id and it builds with no errors and works properly now.  Thanks.

closed 2015-09-17 by jonalling