don/cordova-plugin-ble-central #97: XCode Errors after Updating Plugin

cmivxx, 2015-09-21:
We had your Android version running nicely, however after switching out for IOS and updating your plugin, the following errors pop up in XCode.  Any suggestion how to deal with them?

[image]

don, 2015-09-21:
@cmivxx Xcode 6 or 7? What version of Cordova? What version of the plugin?

I'd try clean then rebuild in Xcode. If that doesn't work, removing and re-installing the iOS platform with Cordova sometimes helps, especially if the project was generated with an older version of Cordova or Xcode. WARNING that will delete any customizations to the native project.

cmivxx, 2015-09-21:
Sorry, I left that part out..   

Xcode 7, Cordova 4.2 and the current Master branch version you have in the repo.

cmivxx, 2015-09-21:
FYI, project builds without issue on android and responds correctly to your BLE plugin.  So...  Good Job?  :)  Now I just need IOS to function and I'm good to go!

cmivxx, 2015-09-21:
Oh yeah, also running Ionic 1.5.5

don, 2015-09-22:
I think it's something in your generated project, that is causing the error. I downgraded to Cordova 4.2

	npm install -g cordova@4.2.0

Try creating a dummy projects and see if you get the same error

	cordova create foo com.example.foo Foo
	cd foo
	cordova plugin add https://github.com/don/cordova-plugin-ble-central
	cordova platform add ios
	open platforms/ios/Foo.xcodeproj

cmivxx, 2015-09-22:
I'll give that a try in just a sec.  Do you thin kit's the older version of Cordova that's causing it?  I'd be happy to try and upgrade that part and see what happens.

BTW: Thanks for the help!

cmivxx, 2015-09-22:
By George that does appear to work.  Well crap, now I have to figure out where the issue lies so I can fix both sides of this.  Fun for me.  Any idea why that might be throwing something off?  I assume I can just likely install the prereq's and try to bring over the project folder and see what happens.

Thanks again for the help.

don, 2015-09-22:
Mine worked with the older Cordova 4.2.0. However, it's possible your Xcode project generated by Cordova is from an even older version?

Xcode doesn't like the extension to CBPeripheral. Not sure why but sometimes it gets in this state.

If `Product -> Clean` then `Product -> Build` doesn't work I'd try regenerating the Xcode project.

    cordova platform rm ios
    cordova platform add ios

cmivxx, 2015-09-22:
That did it...  You are the man!  Thank you for an awesome plugin sir!

closed 2015-09-22 by cmivxx

don, 2015-09-22:
Glad you got it working. What was the final solution? Clean build or regenerate the project?