don/cordova-plugin-ble-central #99: Reading characteristic - only getting one byte

blistick, 2015-09-22:
Hi,

First, thanks for your excellent work on this plugin!

I'm using the plugin in an iOS app that communicates with an Arduino+BLE board called "Bluno". (Bluno integrates with a TI CC2540 BT 4.0 chip with the Arduino UNO development board.)

The iOS app can successfully write to the BLE characteristic, but when it tries to read the characteristic it always gets just one byte, and the value of that byte never changes (even though it is being set by the Bluno).

I'm seeing didUpdateValueForCharacteristic in my console (in Xcode). Should I be seeing something else?

Any clues would be greatly appreciated.

Best,

Barton Listick

don, 2015-09-22:
Can you read the characteristic with [LightBlue](https://itunes.apple.com/us/app/lightblue-bluetooth-low-energy/id557428110?mt=8)?

What does your callback for the read look like?

blistick, 2015-09-22:
Hi Don,

I just tried with LightBlue, and the same sort of thing happens. It reads one byte, with the same value (0x01) each time I ask it to read - and its not the byte that I'm trying to write on the device. However, if I write a byte using LightBlue, it will then subsequently read that same byte. So it seems that the Arduino Serial.write command that's supposed to send bytes to the BLE chip on the Bluno board isn't working, at least not how I understand it.

On a related note, I also tried using your BLE Serial plugin, but its list function returns zero devices. I guess the Bluno doesn't this protocol.

Best,

Barton

don, 2015-09-23:
Try writing with LightBlue and reading with the plugin. If the Bluno service is mimicking SPP, you can try modifying the UUIDs in [RedBearLab](https://github.com/don/cordova-plugin-ble-central/tree/master/examples/redbearlab) or [BluefruitLE](https://github.com/don/cordova-plugin-ble-central/tree/master/examples/bluefruitle) examples to work with your hardware.

[BluetoothSerial](https://github.com/don/BluetoothSerial) works with a fixed set of device. You can add additional ones like Bluno by editing [BLEDefines.h](https://github.com/don/BluetoothSerial/blob/master/src/ios/BLEDefines.h). Note that BluetoothSerial uses Classic on Android where this plugin supports BLE on Android.

blistick, 2015-09-23:
Don, thanks much for these suggestions. I'll try them tomorrow and see what headway I can make.

Most importantly, do you have a tip jar? And/or can I buy you a beer if you've ever in Los Angeles?

don, 2015-09-23:
@blistick I don't have a tip jar. I'll give you a shout if I make it to Los Angeles.

Reopen this issue if you're not able to solve this.

closed 2015-09-23 by don

blistick, 2015-09-23:
Hi Don,

I've made progress on this by using ble.startNotification instead of ble.read.

The reason I didn't use the notification approach before is that I get an error when I use it, BUT I've found that if I show the error with an alert(), AND if I don't dismiss the alert on my phone, I get the notifications and the correct data from my device.

Weird, I know.

Once I dismiss the alert, I stop getting the data, but I do still see "didUpdateValueForCharacteristic" when my device sends data.

If I omit the failure callback from ble.startNotification, or if I do a console.log() within the callback instead of an alert(), the data is not returned by the notification.

See below for my console log. You'll see where I logged bytes that were received, which was happening while the alert was still on my phone.

I'm very close, if only I can get the bytes under all circumstances.

Best,

Barton

2015-09-23 13:32:57.365 Bluno[190:3255] registering for notification
2015-09-23 13:32:57.365 Bluno[190:3255] getData
2015-09-23 13:32:57.366 Bluno[190:3255] Looking for DFB1
2015-09-23 13:32:57.614 Bluno[190:3255] Error Domain=CBATTErrorDomain Code=10 "The attribute could not be found." UserInfo=0x17407ebc0 {NSLocalizedDescription=The attribute could not be found.}
2015-09-23 13:33:00.883 Bluno[190:3255] didUpdateValueForCharacteristic
2015-09-23 13:33:00.890 Bluno[190:3255] Data received from BLE:
2015-09-23 13:33:00.891 Bluno[190:3255] 72
2015-09-23 13:33:00.891 Bluno[190:3255] 69
2015-09-23 13:33:00.892 Bluno[190:3255] 76
2015-09-23 13:33:00.892 Bluno[190:3255] 76
2015-09-23 13:33:00.893 Bluno[190:3255] 79
2015-09-23 13:33:05.294 Bluno[190:3255] didUpdateValueForCharacteristic
2015-09-23 13:33:05.298 Bluno[190:3255] Data received from BLE:
2015-09-23 13:33:05.299 Bluno[190:3255] 72
2015-09-23 13:33:05.299 Bluno[190:3255] 69
2015-09-23 13:33:05.300 Bluno[190:3255] 76
2015-09-23 13:33:05.300 Bluno[190:3255] 76
2015-09-23 13:33:05.300 Bluno[190:3255] 79
2015-09-23 13:33:10.394 Bluno[190:3255] didUpdateValueForCharacteristic
2015-09-23 13:33:15.494 Bluno[190:3255] didUpdateValueForCharacteristic
2015-09-23 13:33:20.324 Bluno[190:3255] didUpdateValueForCharacteristic

don, 2015-09-23:
Based on the [bluno sample](https://github.com/DFRobot/BlunoBasicDemo/blob/feeefd67764e0b3f10fd8381bed3aa868074aa9a/IOS/BlunoBasicDemo/BlunoTest/Bluno/DFBlunoManager.m) notification is the way to go.

What error are you getting when you startNotification?