don/cordova-plugin-ble-central #100: Limited number of startNotifications allowed?

kerrynh, 2015-09-23:
Hi :-)
Thank you so much for this plugin!
I'm having some trouble setting/triggering notifications... I have a service with 20 characteristics, and need to set startNotifications on all of them. Unfortunately, only the first 4 appear to be triggered... I've tried changing the list around, and no matter what order it's in, only the first 4 work.
I'm new to BLE, so I apologise if I'm making a rookie mistake... Is there a limit to the number of startNotifications allowed per service?
Kind regards,
Kerryn

don, 2015-09-23:
@kerrynh I haven't seen this limit, but I usually only have a few notifications.

Are you seeing this on iOS or Android?

kerrynh, 2015-09-23:
I'm currently testing on Android, but I'll see if I can do a build on iOS and let you know...

kerrynh, 2015-09-23:
Ok... All 20 notifications appear to be working correctly on iOS (version 8 - I'm hesitant to upgrade to 9, as other BLEUtility apps we have then cease to work :-s), so it appears that it's an Android issue :-(

As an aside, I also haven't been able to filter on the ble.scan for either of two OSs - I had to call "ble.scan([], 5, onScan, scanFailure);" and check for my device in the onScan function in both cases... (I did see in your examples that you made a note that Android's filtering is broken, but it may be the case for iOS too...)

don, 2015-09-23:
I created a [sample project](https://github.com/don/many-ble-notifications). I'm getting similar OS/hardware dependent results.

Android 5.1.1 on Nexus 5 - All 20 characteristics register. The server sends 20 notifications. Android gets 15. UUID filtering works on Android 5.1.1. I'd expect older hardware and older Android versions work worse, but I haven't tested them yet.

iOS 8.4.1 on iPod Touch (5th gen) gets all 20 notifications.

I haven't had a problem with UUID filtering on iOS. Are you passing a 128 bit or 16 bit UUID to scan?

kerrynh, 2015-09-25:
Thanks Don :-) I'll look at your sample project now...

I checked iOS 9 this morning, and it also gets all 20 notifications. 

I've been playing around with other BLE apps on the Play Store, and I haven't yet found one that has notifications on all 20 (at the same time), and works... For example, using BLEDeng v0.6.8, my tablet (Android 4.3) only registers 9 of the 20 as they change, and my colleague's phone (Android 5.1) registers 15. The read functionality works fine though, as does the read on my app... 

As a workaround, I could perhaps try trigger a read every sec, but I'm not particularly keen to do that, as I think it'll cause performance issues...

Do you have any idea how we might get around this? Or do you think I'll have to try create a native app, as opposed to a hybrid one?

As for the UUID filtering, I'm passing a 128 bit UUID of the form xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx

don, 2015-10-12:
@kerrynh I think this is a hardware limitation, so building a native app won't help either.

Do you have any control over the service? Maybe the limitation is different if you break this up into fewer notifications across more services? (e.g. 4 services with 5 notifications each?)

Polling instead of notifications should work, but will use more battery.

don, 2015-10-21:
Closing issue since this limit is based on the ble hardware.

@kerrynh please let me know if you find a suitable work around

closed 2015-10-21 by don