don/cordova-plugin-ble-central #102: Scanning in iOS 9 fails to find peripheral that is found in iOS 8

JahBrewski, 2015-09-24:
I am using the same code as I previously used in iOS 8 and now my app is not finding my bluetooth peripheral. Is this a known issue? Have other people experienced this?

kerrynh, 2015-09-25:
I tested my app this morning, and I was able to find my peripheral and access the characteristics without a problem in both iOS 8 and 9...
If you've used filtering, perhaps try not doing so? ("ble.scan([], 5, onScan, scanFailure);"), and rather do a check for your peripheral in the onScan function... I couldn't get the filtering to work

JahBrewski, 2015-09-28:
Alright, I've got it working on my phone with iOS 9... but for some reason it isn't working on my client's phone. I've verified that he has bluetooth turned on, he's running the same version of the app I am, etc. What could possibly be the issue?

kerrynh, 2015-09-28:
@JahBrewski, when you say it isn't working, what isn't? Is the app running, but the scan failing? Or is the  connect failing? Are you trying to connect to the device from both apps/phones at the same time? I could be wrong, but I think the device you're trying to connect to, needs to be "free", so ensure that your app has released it before testing with your client's phone.

don, 2015-10-07:
@JahBrewski If it works on you phone but not the client's, I'd expect that @kerrynh has the right idea. The peripheral might be connected to another device so it's not advertising. Rebooting the peripheral usually fixes this.

If you continue to have problems, re-open this issue with more details.

closed 2015-10-07 by don