don/cordova-plugin-ble-central #103: Connecting To BLE Device

kingsleytagbo, 2015-09-28:
Hi:

I am unable to scan for BLE devices on either iMac with Yosemite / Windows.10.

I imported the BLE plugin and referenced it in the PhoneGap App's Index.html App and in the Index.js, I am tried scanning for the BLE devices and I got an error saying that BLE is not enabled even though, I have BLE devices already connected to by Windows 10.1.

What am I doing wrong?

---Index.html 
        <script type="text/javascript" src="phonegap.js"></script>
        <script type="text/javascript" src="ble.js"></script>
        <script type="text/javascript" src="js/index.js"></script>
        <script type="text/javascript">
            app.initialize();
        </script>

--- Index.js

    onDeviceReady: function() {
        app.receivedEvent('deviceready');
        try{
            ble.isEnabled(
                function() {
                    alert("Bluetooth is enabled");
                },
                function() {
                    alert("Bluetooth is *not* enabled");
                }
            );
        }
        catch(e)
        {
            var errorMessage  = "error on isEnabled";
            alert(errorMessage);
        }
        
        try{
            ble.startScan([], function(device) {
                    alert("start scanning blue tooth");
                }, function() { alert("failed to start scan"); });
        }
        catch(e)
        {
            var errorMessage  = "error on startScan";
            alert(errorMessage);
        }
    },

luposlip, 2015-09-28:
BLE central plugin only works on devices, so you have to code your way
around your non-mobile development environment (mock data etc.).

Best,
Henrik

kingsleytagbo, 2015-09-28:
Hi:

Thanks for the feedback. 

It looks like the best option is to build and deploy to a device even for development purposes.

Kingsley

don, 2015-10-07:
@kingsleytagbo you need to develop on a device or use mocks like @luposlip suggests

closed 2015-10-07 by don