don/cordova-plugin-ble-central #104: Using this plugin with "cordova-plugin-bluetooth-serial" on iOS

PeacePan, 2015-10-03:
Hi, don

First,
I add the plugin "cordova-plugin-ble-central", then it's work well on iOS and android.

with my some work, I need use BLE and SPP on android, so I add the plugin "cordova-plugin-bluetooth-serial", but after install, iOS ble function has some problem,  like work on BluetoothSerial, not BLECentralPlugin.

"cordova-plugin-ble-central" and "cordova-plugin-bluetooth-serial" can using together on iOS?

sorry for my poor english.

closed 2015-10-07 by PeacePan

don, 2015-10-07:
You should be OK using both on Android. On iOS, both plugins use Bluetooth Low Energy so you'll need to pick one.

For iOS with BLE, there's no official SPP. If you need SPP-like support on iOS, use cordova-plugin-ble-central to read and write from the UART/SPP service. See the [redbearlab example](https://github.com/don/cordova-plugin-ble-central/tree/master/examples/redbearlab) and [bluefruitle example](https://github.com/don/cordova-plugin-ble-central/tree/master/examples/bluefruitle) for code that calls a SPP-like service.

PeacePan, 2015-10-07:
OK, thank, don

I used plugman to sloved my problem.
Add the BluetoothSerial plugin only android

plugman install --platform android --project <project path>/platforms/android/ --plugin https://github.com/don/BluetoothSerial.git --plugins_dir <project path>/plugins/